gotthardp/lorawan-server #79: MQTT JSON parsing

Sviests, 2017-05-05:
Hi!

Could you give a more detailed explanation on how the Parse Uplink and Build Downlink options work? Erlang is completely new to me, so that might be the reason why it doesn`t seem obvious to me.
What I`m trying to do is parse the JSON data that the server posts to MQTT so that I could post specific values to specific MQTT topics, for example, the battery value would be posted on a topic {devaddr}/battery, payload would be posted to {devaddr}/payload etc... Hopefully this is possible...

Currently the format of messages posted to MQTT is:
`{"shall_reply":true,"last_lost":false,"data":"10000000B224004C00","port":24,"fcnt":192,"fields":null,"rxq":{"lsnr":9.8,"rssi":-48,"tmst":300757724,"codr":"4/5","datr":"SF12BW125","freq":868.1},"datetime":"2017-05-03T10:26:15Z","appargs":null,"devaddr":"XXXXXXX"}`

gotthardp, 2017-05-05:
You need to write a function as described at https://github.com/gotthardp/lorawan-server/blob/master/doc/Backends.md
```erlang
fun(_Port, <<LED, Press:16, Temp:16, AltBar:16, Batt, Lat:24, Lon:24, AltGps:16>>) ->
  #{led => LED, pressure => Press, temp => Temp/100, alt_bar => AltBar, batt => Batt}
end.
```

It's based on pattern matching. The `<<A, B, C>>` is a binary pattern, where A, B, C are "variables" corresponding to the values encoded in the binary. Erlang matches the binary data against this pattern and fills the "variables" with the value in the binary. By default its 8 bits. By saying `Temp:16` you say it's 16-bits, big endian.
Full syntax is described at http://erlang.org/doc/programming_examples/bit_syntax.html#id66791

Once you have the variables, you put them in a structure that gets encoded in JSON. The `#{name1 => A, name2 => B, name3 => C}` creates a JSON `{"name1":A, "name2":B, "name3":C}`.

Shall I described in even in more details?

Sviests, 2017-05-05:
If I understand it correctly, to get only the "data" field from my example I could have something like this as my uplink parser:
`fun(_Port, <<Data:12>>) ->
   #{data => Data}
end.`

gotthardp, 2017-05-05:
Exactly. The length has to match the payload, which is a multiply of 8-bits.
The following takes 2-byte long payload and puts it as integer to the {"data":xy} field.
```
fun(_Port, <<Data:16>>) -> #{data => Data} end.
```

Sviests, 2017-05-05:
Unfortunately this does not work for me... I am sure that the handler is set up properly as it does post the received messages to the MQTT topic, but I receive exactly the same JSON data as before...
Sucks that I can test it only once an hour because that`s the default reporting rate of my sensor, when I`ll figure out how to send downlink messages, I`ll be able to set a shorter interval...

gotthardp, 2017-05-05:
Does it print some error?

Sviests, 2017-05-05:
No, it acts like nothing happened.. This is how my Handler look like:
[image]
This is data posted to MQTT (before parser and after):
[image]

gotthardp, 2017-05-05:
A 128-bit number? That may be too large to handle.

Sviests, 2017-05-05:
The payload that I am receiving from my sensor is a byte array, 9 bytes in size, I tried using 82 for size, that did not make a change either...

gotthardp, 2017-05-05:
It has to convert something, or write an error to the log. If it doesn't do anything, then the function is not invoked. Did you set device/node group name?

For parsing arrays it's better to do
```erlang
fun(_Port, Bin) ->
#{data => binary_to_list(Bin)}
end.
```

this will put the entire array to the payload (but please note it is already there in the "data" field)

Sviests, 2017-05-05:
I just found something in the error.log:
` <0.8370.0>@lorawan_application_backend:data_to_fields:116 Fun failed error:function_clause`

I will try the function you suggested.

Sviests, 2017-05-05:
Ok, I think I got it working, but I've misunderstood the functionality of it, I see that it puts the value in the `fields` field of the JSON data I am receiving on my MQTT topic, I was hoping that I could get just the `data` field and the rest of the fields filtered out...

Regarding the downlink messages, could you please provide an example of how I could send a downlink message that contains a payload using MQTT. 

On my gateway, I am using the packet_forwarder code from Semtech if that makes any difference...

gotthardp, 2017-05-05:
Would the filtering function be useful to you? It's not that big amount of data, is it? Different people requested different fields, so this is why it became so huge.

The downlinks work the same way as in https://github.com/gotthardp/lorawan-server/blob/master/doc/WebSockets.md, so you send something like `{"devaddr":"11223344", "data":"0026BF08BD03CD35000000000000FFFF", "confirmed":true}`.

Sviests, 2017-05-05:
Well, what I`m trying to make is a system that collects readings from sensors (water meter, electricity meter, etc...) and sends the information to a monitoring server (through your LoraWAN server) and visualises this data, generates alarms when certain events are triggered etc... Most of the field posted to MQTT are not necessary for me in this case as the payload is what really matters, the rest of the data to me is only necessary when I open up the Web UI of your LoraWAN server.

A filtering feature would be nice, but if I`m the only one asking for it, it might not be that necessary :D

gotthardp, 2017-05-05:
I will:
 - make the JSON elements filterable
 - document the downlink and the binary filtering a bit more

Sviests, 2017-05-05:
Thank you very much! This will be very helpful!

gotthardp, 2017-05-10:
The improved documentation is already available. The filterable JSON is planned for the release 0.5.0 and will be tracked by issue #86.

closed 2017-05-10 by gotthardp