gotthardp/lorawan-server #82: OTAA, device sent JOIN request, nothing more.

mrodesign, 2017-05-08:
Hello, 
I have Lorank8 with my own devices with RN2483 Microchip 868 modules.
I use them with public TTN. It works fine.
Today I need to create private Lora network, so I tried to use gotthardp/lorawan-server
I installed everything according docs. I started served with debug option. I changed global_conf.json in poly_packet_fwd on Lorank8 to send data to localhost:1680.
Next, I configured server by web interface - add gateway and device.
In console I see:
`12:18:45.878 [info] gateway 1DEE01CA00D033D5 {connected,{{127,0,0,1},57641}}
`
and:
```
12:18:52.145 [info] packet_forwarder interface terminated: {function_clause,[{lorawan_gw_forwarder,'-handle_info/2-fun-0-',[{time,<<50,48,49,55,45,48,53,45,48,56,84,49,50,58,49,56,58,53,50,90>>}],[{file,[47,104,111,109,101,47,116,114,97,118,105,115,47,98,117,105,108,100,47,103,111,116,116,104,97,114,100,112,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,102,111,114,119,97,114,100,101,114,46,101,114,108]},{line,55}]},{lists,foreach,2,[{file,[108,105,115,116,115,46,101,114,108]},{line,1338}]},{lorawan_gw_forwarder,handle_info,2,[{file,[47,104,111,109,101,47,116,114,97,118,105,115,47,98,117,105,108,100,47,103,111,116,116,104,97,114,100,112,47,108,111,114,97,119,97,110,45,115,101,114,118,101,114,47,95,98,117,105,108,100,47,100,101,102,97,117,108,116,47,108,105,98,47,108,111,114,97,119,97,110,95,115,101,114,118,101,114,47,115,114,99,47,108,111,114,97,119,97,110,95,103,119,95,102,111,114,119,97,114,100,101,114,46,101,114,108]},{line,54}]},{gen_server,try_dispatch,4,[{file,[103,101,110,95,115,101,114,118,101,114,46,101,114,108]},{line,601}]},{gen_server,handle_msg,5,[{file,[103,101,110,95,115,101,114,118,101,114,46,101,114,108]},{line,667}]},{proc_lib,init_p_do_apply,3,[{file,[112,114,111,99,95,108,105,98,46,101,114,108]},{line,247}]}]}
12:18:52.148 [error] gen_server lorawan_gw_forwarder terminated with reason: no function clause matching lorawan_gw_forwarder:'-handle_info/2-fun-0-'({time,<<"2017-05-08T12:18:52Z">>}) line 55
12:18:52.153 [error] CRASH REPORT Process <0.240.0> with 0 neighbours exited with reason: no function clause matching lorawan_gw_forwarder:'-handle_info/2-fun-0-'({time,<<"2017-05-08T12:18:52Z">>}) line 55 in gen_server:terminate/7 line 812
12:18:52.162 [error] Supervisor lorawan_sup had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.240.0> exit with reason no function clause matching lorawan_gw_forwarder:'-handle_info/2-fun-0-'({time,<<"2017-05-08T12:18:52Z">>}) line 55 in context child_terminated
12:18:52.212 [debug] Supervisor lorawan_sup started lorawan_gw_forwarder:start_link([{port,1680}]) at pid <0.242.0>
12:18:52.295 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<118,32,205,64>>
12:18:56.204 [warning] gateway 1DEE01CA00D033D5 {downlinks_lost,1}
```

Could anyone help me, what I did wrong?
Krzysztof

gotthardp, 2017-05-08:
You did nothing wrong. I see two problems there:

1) Your gateway sends unexpected data `{time,<<"2017-05-08T12:18:52Z">>}`. Do you have some special build of the gateway? Is there  any reason why the gateway sends this undocumented field?

2) Your gateway didn't like the downlink frame sent from the server. (Could be due to a bug in the server.) Could you please have a look in the gateway log and see if there is some `ERROR` report?

mrodesign, 2017-05-09:
Hi,
I use standard poly_pkt_fwd embedded in Lorank8. (https://webshop.ideetron.nl/Files/3/1000/1211/Attachments/Product/23t415f609w49D376q7N237h696Q544S.pdf)  Have you ever tried your server on that device? 
To check what this pkt_fwd sends I connected to local node-red upd:
[image]
I

gotthardp, 2017-05-09:
Users reported that lorank8 is (or was) compatible. The problem is a proprietary message format used by lorank8. In your gateway config you should have "stats format" set to "ideetron concise" (or so). When you set this to "semtech" you should get this work.

The lorawan-server doesn't (for now) support the lorank8 proprietary extensions.

mrodesign, 2017-05-11:
I changed "stats format" to "semtech" but nothing changes.

gotthardp, 2017-05-11:
And what is written in the logs now?

mrodesign, 2017-05-11:
`06:49:28.148 [error] gen_server lorawan_gw_forwarder terminated with reason: no function clause matching lorawan_gw_forwarder:'-handle_info/2-fun-0-'({time,<<"2017-05-11T06:49:28Z">>}) line 55
06:49:28.154 [error] CRASH REPORT Process <0.374.0> with 0 neighbours exited with reason: no function clause matching lorawan_gw_forwarder:'-handle_info/2-fun-0-'({time,<<"2017-05-11T06:49:28Z">>}) line 55 in gen_server:terminate/7 line 812
06:49:28.160 [error] Supervisor lorawan_sup had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.374.0> exit with reason no function clause matching lorawan_gw_forwarder:'-handle_info/2-fun-0-'({time,<<"2017-05-11T06:49:28Z">>}) line 55 in context child_terminated`

The same situation.

gotthardp, 2017-05-11:
I see. They introduced one more non-standard field that cannot be disabled. I made a modification to the sources. Would you be able to download latest `master` and try that, please?

mrodesign, 2017-05-11:
OK, you did it well :) I don't have any errors on server.

But I can't still JOIN my end device.
[image]

```
07:36:05.646 [info] gateway 1DEE01CA00D033D5 {connected,{{127,0,0,1},43921}}
07:36:31.387 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
07:37:16.947 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
```

gotthardp, 2017-05-11:
Could you please have a look in the gateway log if:
 * it indicates it received an uplink
 * it indicates any error or warning

The downlink direction is always a pain to debug as there is so many different parameters to set  and the gateway config must be aligned with what the server thinks.

mrodesign, 2017-05-11:
```
##### 2017-05-11 09:44:35 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 2 (46 bytes)
# PUSH_DATA datagrams sent: 2 (470 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 2 (354 bytes)
# RF packets sent to concentrator: 2 (34 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:2, rej:0)
# TX rejected (collision beacon): 0.00% (req:2, rej:0)
# TX rejected (too late): 0.00% (req:2, rej:0)
# TX rejected (too early): 0.00% (req:2, rej:0)
### BEACON IS DISABLED! 
### [JIT] ###
# INFO: JIT queue contains 0 packets.
# INFO: JIT queue contains 0 beacons.
### GPS IS DISABLED! 
### [PERFORMANCE] ###
# Upstream radio packet quality: 100.00%.
# Upstream datagram acknowledgment quality for server "127.0.0.1" is 100.00%.
# Downstream heart beat acknowledgment quality for server "127.0.0.1" is 100.00%.
# Downstream datagram content quality for server "127.0.0.1" is 100.00%.
# Downstream radio transmission quality for server "127.0.0.1" is 100.00%.
##### END #####
```

This is my log from GW

gotthardp, 2017-05-11:
This looks OK. In that case the problem is that the device listens are different frequency or at different time than the server. Is there anything you can configure in the device (time-slots, frequencies)?

mrodesign, 2017-05-11:
Has server specific frequency or SF to join?
My configuration on end-node works correctly for TTN and Brocaar's server.
A few time ago there was a problem with TTN. Devices joins only on SF9.

gotthardp, 2017-05-11:
I believe my server is standard compliant, but from time to time there is a configuration issue that breaks the downlinks. Would you please be able to use wireshark or `sudo netstat -uapn` to capture the server to gateway UDP communication both for the TTN that works as well as for the server that doesn't work?

mrodesign, 2017-05-11:
```
udp        0      0 127.0.0.1:41750         127.0.0.1:1680          ESTABLISHED 1724/poly_pkt_fwd
udp        0      0 127.0.0.1:58653         127.0.0.1:1680          ESTABLISHED 1790/poly_pkt_fwd
udp        0      0 127.0.0.1:54828         127.0.0.1:1680          ESTABLISHED 1724/poly_pkt_fwd
udp        0      0 192.168.1.123:44849     52.169.76.203:1700      ESTABLISHED 1790/poly_pkt_fwd
udp        0      0 127.0.0.1:53            0.0.0.0:*                           477/connmand    
udp        0      0 0.0.0.0:44093           0.0.0.0:*                           247/systemd-timesyn
udp        0      0 0.0.0.0:37694           0.0.0.0:*                           438/avahi-daemon: r
udp        0      0 192.168.1.123:42825     192.168.1.1:53          ESTABLISHED 477/connmand    
udp        0      0 192.168.1.123:36445     52.169.76.203:1700      ESTABLISHED 1790/poly_pkt_fwd
udp        0      0 192.168.1.123:53604     192.168.1.1:53          ESTABLISHED 477/connmand    
udp        0      0 127.0.0.1:38762         127.0.0.1:1680          ESTABLISHED 1790/poly_pkt_fwd
udp        0      0 0.0.0.0:1680            0.0.0.0:*                           1622/beam       
udp        0      0 127.0.0.1:54187         127.0.0.1:53            ESTABLISHED 477/connmand    
udp        0      0 192.168.1.123:48056     192.168.1.1:53          ESTABLISHED 477/connmand    
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           438/avahi-daemon: r
udp6       0      0 :::41986                :::*                                438/avahi-daemon: r
udp6       0      0 ::1:59421               ::1:53                  ESTABLISHED 477/connmand    
udp6       0      0 ::1:53                  :::*                                477/connmand    
udp6       0      0 :::5353                 :::*                                438/avahi-daemon: r
```

gotthardp, 2017-05-11:
Oops. Sorry, I meant to do `sudo tcpdump -i lo -w file.pcap` and send me the `file.pcap`? If there is no tcpdump you should `sudo apt-get install tcpdump`.

mrodesign, 2017-05-11:
This is for local serv:
[file.pcap.zip](https://github.com/gotthardp/lorawan-server/files/992961/file.pcap.zip)

gotthardp, 2017-05-11:
Thanks a lot for this. The first difference between the two is that the TTN sends the accept in the RX2 window, while I was using RX1 only.

I read in a blog that some Microchip devices use RX2 only, which may be your case. Therefore, I made an update to retransmit JoinAccept in RX2. If this was the problem, the join will work for you. Please try :)

mrodesign, 2017-05-12:
```
05:32:26.314 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:32:26.315 [debug] Retransmitting JoinAccept in RX2
05:33:14.576 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:33:14.577 [debug] Retransmitting JoinAccept in RX2
05:33:45.617 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:33:45.619 [debug] Retransmitting JoinAccept in RX2
05:34:52.580 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:34:52.586 [debug] Retransmitting JoinAccept in RX2
05:36:27.122 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:36:27.124 [debug] Retransmitting JoinAccept in RX2
05:36:50.613 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:36:50.615 [debug] Retransmitting JoinAccept in RX2
05:37:38.518 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:37:38.520 [debug] Retransmitting JoinAccept in RX2
05:38:07.296 [info] JOIN REQUEST <<112,179,213,126,208,0,5,54>> <<0,0,0,0,0,0,0,68>> -> <<86,148,246,180>>
05:38:07.298 [debug] Retransmitting JoinAccept in RX2
```

[file.pcap 3.zip](https://github.com/gotthardp/lorawan-server/files/995129/file.pcap.3.zip)

Unfortunately, there is still problem. I attach log.

gotthardp, 2017-05-13:
Would you be able to add some debug output to the device to see if
 * it is not receiving the Join Accept message, or
 * it does not like the Join Accept it got

The last difference is that TTN sends a list of additional frequencies, whereas my server doesn't. This list of optional, so the device should accept a message without this list. However, if your device is trying to use the channel 4 and beyond, there may be some collision.

mrodesign, 2017-05-15:
A node joins correctly now, we get "accepted" status. Next problem is sending frame with ACK. Node sends data frame, but does not receive ACK. After it several tries device resets, joins correctly again, but again is not able to successfully send data frame with ACK
we use channels 0-4

gotthardp, 2017-05-15:
This looks like a TX1/TX2 issue. I heard there are devices listening on TX2 window only. Couldn't your be one of those? Did you need to do some special config when attaching to TTN?

mrodesign, 2017-05-15:
Hmm, I don't know if it is TX1 or TX2. 
But I don't need special configuration both on TTN and Brocaar's.

gotthardp, 2017-05-15:
What device config do you use? [Some forums](https://www.thethingsnetwork.org/forum/t/ttn-uno-beta-release-documentation/290/41) recommend `mac set rx2 3 869525000` which will definitely not work with my server. You should better use `mac set rx2 0 869525000`.

mrodesign, 2017-05-15:
I set as you wrote:
mac set rx2 0 869525000
Nothing better.

gotthardp, 2017-05-15:
The ACK usually comes in RX1. Are you sure that your device listens in RX1?

mrodesign, 2017-05-15:
The RN2483 stack takes care of all the timing and the frequency of the RX1 & RX2 windows. Although yes, you should ensure that the RX2 settings match that of the server. The defaults for EU are 869.525 MHz / DR0 (SF12, 125 kHz).
http://www.microchip.com/forums/m946931.aspx

gotthardp, 2017-05-15:
To debug this further I'd need you do investigate in what window the device listens to the ACK and why it does not receive it. I believe my server is standard compliant. It works well with other devices, even the ACKs.

gotthardp, 2017-05-29:
The 0.4.3 will contain provisions to switch to RX2 only. I am closing this for now, feel free to open another issue when you experience similar issues with the latest release.

closed 2017-05-29 by gotthardp