gotthardp/lorawan-server #83: Backend tab seems missing in the Web Admin pages for me

gfwilliams, 2017-05-09:
Hi, thanks for this! It looks like just what I'm after.

I've just been trying to get this working on a Pi with the ic880a, and am almost there. I have working reception of packets from an RN2483 (via the `Links` tab), but I'm just trying to figure out how to send data.

I figured the best way was to use the MQTT backend and to then send data via MQTT, but I tried to do what's suggested here: 

https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md#generic-mqtt-server

and I don't see a `Backend` menu item in the Server Admin page:

[image]

Any idea what I'm doing wrong? `backend` seems to be in the configuration file for 0.4 (it's not for 0.3 and adding it causes lorawan-server to crash) but it's not available for me on either version.

Also all the documentation also mentions `Nodes`, but the menu items seem to be titled `Links` now? Also in  the 0.4.0 release, `Links` and `Ignored Links` just get a `Not Found` page now. 0.3.0 worked fine for me.

Thanks!

gfwilliams, 2017-05-09:
Just to add I installed from the tar.gz if that helps - I'd started with 0.3 and there was no `.deb` for that. I installed 0.4 the same way, in its own directory. There are no errors or crashes logged in 0.4 (apart from the initial 'unknown mac' before I added the gateway)

gotthardp, 2017-05-09:
What you see is an old menu, probably from 0.3.0. This can happen:
 * when the page is cached somewhere-- refreshing the page and clearing the browser cache usually helps
 * when the old version is still running, e.g. as a daemon

Did you try refreshing the page?

gfwilliams, 2017-05-10:
Thanks! I did try refreshing, yes - but no luck. And it definitely wasn't running the old version too as trying to load the page between running the two versions running would get no HTTP response.

Erlang wouldn't be cacheing pages anywhere global would it? I was running from a totally new directory.

I also tried copying the `nodes/list` URL out of one of your screenshots in the hope I might be able to get to the page directly if the menu was broken, but no luck there either  - just `Not Found`, but no errors reported.

gotthardp, 2017-05-10:
I tried to unpack and run the 0.4.0.tar.gz and it works as it should. There is no "Links" in the entire code, so I am almost sure you have a 0.3.0 version installed somewhere. Just to doublecheck: in the unzipped directory there:
 * must be "releases/0.4.0" only. No releases/0.3.0.
 * must be only one version for each component, i.e. there must be `lorawan_server-0.4.0` only, no `lorawan_server-0.3.0` etc.

The .deb installs under `/usr/lib/lorawan-server`. You are sure there is nothing else too?

gfwilliams, 2017-05-10:
Thanks for testing it - I'll have another go and will let you know. I'll also try forcing Chrome to do a full (non-cached) reload.

gfwilliams, 2017-05-11:
Hi, Sorry for wasting your time on this - I just tried again and it works perfectly. Same folder as before, same everything - just this time it works perfectly. It's strange as I'd tried it multiple times before.

There must have been some kind of caching issue somewhere. Thanks for checking out the 0.4 release again for me though!

closed 2017-05-11 by gfwilliams

gotthardp, 2017-05-11:
No problem. I am glad it works now.