gotthardp/lorawan-server #86: Handlers shall allow enabling/disabling JSON elements

gotthardp, 2017-05-10:
Couple of JSON improvements:

Handler admin shall have checkboxes to enable/disable individual JSON elements. Requested by @Sviests in issue #79.

Handler shall (optionally) include gateway MAC in the JSON uplink data. Requested by @matiasz8 in issue #60.

Sviests, 2017-05-11:
Great news! Can`t wait for this to be implemented. Thanks!!

DigitalPigeon, 2017-05-13:
As per my second comment in #92, could we also optionally include the mac (devEUI) in the JSON as well please?

gotthardp, 2017-05-13:
@DigitalPigeon Yes. Will do.

gotthardp, 2017-06-10:
I just implemented everything. Available now in the `master` branch. By the end of next week will be released as v0.4.5. Please give it a try!

gotthardp, 2017-06-24:
Done in 0.4.5.

closed 2017-06-24 by gotthardp