gotthardp/lorawan-server #88: server could not start

batulzii, 2017-05-12:
C:\lorawan-server\_build\default\rel\lorawan-server\bin>lorawan-server
{"init terminating in do_boot",{undef,[{lorawan_app,start,[],[]},{init,start_em,
1,[{file,"init.erl"},{line,1076}]},{init,do_boot,3,[{file,"init.erl"},{line,784}
]}]}}
init terminating in do_boot ()

Crash dump is being written to: erl_crash.dump...done

C:\lorawan-server\_build\default\rel\lorawan-server\bin>

batulzii, 2017-05-12:
It looks like need to upgrade current Erlang/OTP installation

batulzii, 2017-05-12:
installed Erlang/OTP 19.3, but server crashed.

gotthardp, 2017-05-12:
How? What did it say?

batulzii, 2017-05-12:
my previous Erlang installation was 18, then i uninstall it and install new Erlang/OTP. It is v20 now. compiled server and send to windows. Windows PC has installed v19. And then server could not start just crashed. back to ubuntu and try to remove v20 and install v19.2. but could not remove v20 now. Try to use Kerl  to build and install v19.2

batulzii, 2017-05-12:
there no binary v20 for Windows. Server starting normaly in Ubuntu (erlang/otp v20)

gotthardp, 2017-05-12:
The latest release is 19.3. I never tried the server against 20.0-rc1, which was released just 7 days ago.

batulzii, 2017-05-12:
upgrade failed now with error 
Rebar dependency ssl could not be loaded for reason
when i try:
make upgrade

batulzii, 2017-05-12:
So i could not compile server yet. Do you encounter this ?

cd lorawan-server
make upgrade
:
:
Rebar dependency ssl could not be loaded for reason.....

gotthardp, 2017-05-12:
This appears to be an Erlang or Rebar installation problem. Did you install all Erlang packages?

closed 2017-05-13 by batulzii