gotthardp/lorawan-server #89: No Backend MQTT activity with nothing in logs

DigitalPigeon, 2017-05-12:
Trying to get MQTT connection working, but group device seem to not be triggering the Handler, nor are any incoming messages being captured.

My node is configured as

Application: **backend**
Group: **Demo**

My handler is configured as
Group: **Demo**
Format: **Raw Data**
Connector: **Demo Connector**

My demo connector as
Connector Name: **Demo Connector**
Enabled: **True**
URI: **mqtt://esmqdev1:8883** (this is an internal MQTT broker, running on websphere telemetry)
Published Topic: **mqtt/zedimote/in/{devaddr}**
Subscribe: **mqtt/zedimote/out/#**
Consumed Topic: **mqtt/zedimote/out/{devaddr}**

When frames are received, they are logged as being in group **Demo**

However, no activity with MQTT occurred. The only info in the debug log occurs on enabling the Connector

```
2017-05-12 10:09:19.722 [debug] <0.4311.0>@lorawan_connector_mqtt:init:22 Connecting Demo Connector to mqtt://esmqdev1:8883
2017-05-12 10:09:19.725 [debug] <0.256.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1",8883,"/",[]}, {connector,<<"Demo Connector">>,true,<<"mqtt://esmqdev1:8883">>,<<"mqtt/zedimote/in/{devaddr}">>,...}) at pid <0.4311.0>
```

I'm sure I am doing something wrong, but I'm surprised there is no activity at all in the debug logs if I have.

So in addition to looking for some general help on why this might be happening, possibly a feature request to come out of this would be some more informative logging to help troubleshoot when an MQTT configuration is incorrect.

Thoughts?

DigitalPigeon, 2017-05-12:
Alright, I feel like a bit of a dolt
Our MQTT runs on 1883, not 8883...

A suggestion would be that it would be nice if the log would have indicated it couldn't connect to the misconfigured URI.

Regardless, this did not yet solve the problem. I now see the following in the lora server's **console output** (not log files) every couple of seconds

`[warning] [Client <0.4637.0>] Connection lost for: tcp_closed`

And no data goes in our out of MQTT

I am connecting to the MQTT server using the exact same credentials with a test client, and have no problems publishing or subscribing using that.

gotthardp, 2017-05-12:
1) yes, I should probably add some more logs. I accept the request

2) what kind of MQTT server are you using?

DigitalPigeon, 2017-05-12:
Alright, I think possibly this is an MQTT version issue.

Noticed this in the broker
`May 12, 2017 12:55:44 PM MDT[ServerWorker0] com.ibm.mq.MQXRService.MQTTServerContext
AMQXR0023I: Unsupported MQTT protocol version on channel PlainText, the exception com.ibm.mq.MQXRService.MQTTInvalidVersionException: AMQXR1004E:an invalid message version=4(byte) subVersion=2(byte)was received.  was thrown.`

I can reproduce this using a test client when switching between MQTT 3.1 (working), and 3.1.1. (not working).

It looks like the MQTT brokers we currently have deployed are not compatible with 3.1.1 (these are Websphere MQ 7.5... MQTT 3.1.1 was added in 8.0). So while we will work with our infrastructure teams to look at having those updated to 8 or 9, it will take quite a while I'm sure.

In the mean time, is there any way to get the emqtt client that lorawan-server uses to use 3.1 instead of 3.1.1 by chance?

gotthardp, 2017-05-12:
Many thanks for debugging this.

Yes, I will implement an automatic dowgrade to 3.1 upon a protocol version error.

DigitalPigeon, 2017-05-12:
Thank you very much Petr! Really appreciate the work you have done here.

gotthardp, 2017-05-12:
It turned out that the server kept trying to connect forever, so it never reported an issue. I fixed that and also implemented a fallback to 3.1. Could you please `git pull` latest `master` and check how it works?

DigitalPigeon, 2017-05-12:
Wow! Thanks for the quick turn around.
I'm actually running off of binaries right now so will have to switch over to running off source. I'll let you know how it goes.

Sent from my Bell Samsung device over Canada's largest network.

DigitalPigeon, 2017-05-13:
Well, I am stuck on windows, and unfortunately can't get lorawan-server to build in that environment.
I've simply placed rebar3 within the lorawan-server git root, and get the following (using gnuwin32make)

```
C:\projects\git\lorawan-server>make upgrade 
process_begin: CreateProcess(NULL, env escript C:\projects\git\lorawan-server\rebar3 upgrade, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [upgrade] Error 2
```

I'm not really familiar enough with this stack to troubleshoot much... so I'm probably stuck waiting for the next release build unless the above error is obvious to you.

gotthardp, 2017-05-13:
I see. I built the binary for you. Try downloading http://swimgate.eu/lorawan-server-0.4.0.tar.gz

DigitalPigeon, 2017-05-13:
I deployed the binaries, and the max retries for connecting to the broker works now, but it doesn't seem to be falling back to 3.1

From the log file
```
10:06:51.426 ←[0;38m[debug] Connecting DemoConnector to mqtt://esmqdev1.zedisolutions.com:1883←[0m
10:06:51.427 ←[0;38m[debug] Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at pid <0.3212.0>←[0m
10:07:51.462 ←[1;31m[error] Connector DemoConnector terminated: tcp_closed←[0m
```

From the console
```
[warning] [Client <0.3213.0>] Connection lost for: tcp_closed
[warning] [Client <0.3213.0>] Connection lost for: tcp_closed
[warning] [Client <0.3213.0>] Connection lost for: tcp_closed
[warning] [Client <0.3213.0>] Connection lost for: tcp_closed
[warning] [Client <0.3213.0>] Connection lost for: tcp_closed
[error] [Client <0.3213.0>] reconect error: retries_exhausted
```

Our IT team will be deploying the new broker in our dev environment at the end of next week, that will support 3.1.1... so please don't worry about me too much. But if you want help to try and get this fallback working I'm happy to keep helping you test it out.

gotthardp, 2017-05-13:
Thanks. I would like to fix it anyway as other users may have the same problem. Could test it again, please with the new binary http://swimgate.eu/lorawan-server-0.4.0.tar.gz

I just added more MQTT logs.

DigitalPigeon, 2017-05-13:
Console:
```
10:50:35.433 ←[0;38m[debug] Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at pid <0.245.0>←[0m
[debug] [lora1@192.168.128.128:60193] SENT: CONNECT(Q0, R0, D0, ClientId=lora1,ProtoName=MQTT, ProtoVsn=4, CleanSess=true, KeepAlive=0, Username=undefined, Password=undefined)
[debug] [lora1@192.168.128.128:60193] SENT: <<16,17,0,4,77,81,84,84,4,2,0,0,0,5,108,111,114,97,49>>
[info] [Client <0.246.0>] connected with esmqdev1.zedisolutions.com:1883
[warning] [Client <0.246.0>] Connection lost for: tcp_closed
[info] [Client <0.246.0>] try reconnecting...
[info] [Client <0.246.0>]: connecting to esmqdev1.zedisolutions.com:1883
[debug] [lora1@192.168.128.128:60199] SENT: CONNECT(Q0, R0, D0, ClientId=lora1,ProtoName=MQTT, ProtoVsn=4, CleanSess=true, KeepAlive=0, Username=undefined, Password=undefined)
[debug] [lora1@192.168.128.128:60199] SENT: <<16,17,0,4,77,81,84,84,4,2,0,0,0,5,108,111,114,97,49>>
[info] [Client <0.246.0>] connected with esmqdev1.zedisolutions.com:1883
[warning] [Client <0.246.0>] Connection lost for: tcp_closed
[info] [Client <0.246.0>] try reconnecting...
[info] [Client <0.246.0>]: connecting to esmqdev1.zedisolutions.com:1883
[debug] [lora1@192.168.128.128:60201] SENT: CONNECT(Q0, R0, D0, ClientId=lora1,ProtoName=MQTT, ProtoVsn=4, CleanSess=true, KeepAlive=0, Username=undefined, Password=undefined)
[debug] [lora1@192.168.128.128:60201] SENT: <<16,17,0,4,77,81,84,84,4,2,0,0,0,5,108,111,114,97,49>>
[info] [Client <0.246.0>] connected with esmqdev1.zedisolutions.com:1883
[warning] [Client <0.246.0>] Connection lost for: tcp_closed
[info] [Client <0.246.0>] try reconnecting...
[info] [Client <0.246.0>]: connecting to esmqdev1.zedisolutions.com:1883
[debug] [lora1@192.168.128.128:60206] SENT: CONNECT(Q0, R0, D0, ClientId=lora1,ProtoName=MQTT, ProtoVsn=4, CleanSess=true, KeepAlive=0, Username=undefined, Password=undefined)
[debug] [lora1@192.168.128.128:60206] SENT: <<16,17,0,4,77,81,84,84,4,2,0,0,0,5,108,111,114,97,49>>
[info] [Client <0.246.0>] connected with esmqdev1.zedisolutions.com:1883
[warning] [Client <0.246.0>] Connection lost for: tcp_closed
[info] [Client <0.246.0>] try reconnecting...
[info] [Client <0.246.0>]: connecting to esmqdev1.zedisolutions.com:1883
[debug] [lora1@192.168.128.128:60208] SENT: CONNECT(Q0, R0, D0, ClientId=lora1,ProtoName=MQTT, ProtoVsn=4, CleanSess=true, KeepAlive=0, Username=undefined, Password=undefined)
[debug] [lora1@192.168.128.128:60208] SENT: <<16,17,0,4,77,81,84,84,4,2,0,0,0,5,108,111,114,97,49>>
[info] [Client <0.246.0>] connected with esmqdev1.zedisolutions.com:1883
[warning] [Client <0.246.0>] Connection lost for: tcp_closed
[info] [Client <0.246.0>] try reconnecting...
[error] [Client <0.246.0>] reconect error: retries_exhausted
```

For reference, here are the errors in the same period from the broker
```
--------------------------------------------------------------------
May 13, 2017 10:50:35 AM MDT[ServerWorker0] com.ibm.mq.MQXRService.MQTTServerContext
AMQXR0023I: Unsupported MQTT protocol version on channel PlainText, the exception com.ibm.mq.MQXRService.MQTTInvalidVersionException: AMQXR1004E:an invalid message version=4(byte) subVersion=2(byte)was received.  was thrown.
--------------------------------------------------------------------
May 13, 2017 10:51:35 AM MDT[ServerWorker0] com.ibm.ws.objectManager.utils.TraceFactoryImpl$SuppressedMessage
AMQCO2004I ''3'' instances of message ''AMQXR0023I:'' were suppressed.
--------------------------------------------------------------------
May 13, 2017 10:51:35 AM MDT[ServerWorker0] com.ibm.mq.MQXRService.MQTTServerContext
AMQXR0023I: Unsupported MQTT protocol version on channel PlainText, the exception com.ibm.mq.MQXRService.MQTTInvalidVersionException: AMQXR1004E:an invalid message version=4(byte) subVersion=2(byte)was received.  was thrown.
--------------------------------------------------------------------
```

If its at all helpful, here is an example of Mosquitto_sub working against this broker under 31, but getting the same connection drop as you at 311

```

C:\Program Files (x86)\mosquitto>mosquitto_pub.exe -d -h esmqdev1 -t mqtt/test -i lora2 -m "Hello World" -V mqttv31
Client lora2 sending CONNECT
Client lora2 received CONNACK
Client lora2 sending PUBLISH (d0, q0, r0, m1, 'mqtt/test', ... (11 bytes))
Client lora2 sending DISCONNECT

C:\Program Files (x86)\mosquitto>mosquitto_pub.exe -d -h esmqdev1 -t mqtt/test -i lora2 -m "Hello World" -V mqttv311
Client lora2 sending CONNECT
Error: The connection was lost.
```

gotthardp, 2017-05-13:
Thanks a bunch! I thought the server will respond with a specific error message. It is just closing the connection. Grrrr.

DigitalPigeon, 2017-05-13:
Yeah, I was afraid of that.
I know its a lot more work (UI changes, data model changes), but the most amazing solution would to be able to specify the mqtt protocol version when configuring the connector.
I understand that might be something you're interested in adding however.

gotthardp, 2017-05-13:
I was hoping the connector will be generic. Adding a version is possible, but against the intended philosophy, which makes me unhappy. :)

gotthardp, 2017-05-13:
Would you mind doing another attempt with http://swimgate.eu/lorawan-server-0.4.0.tar.gz, please?

DigitalPigeon, 2017-05-13:
Interesting, there is now a crash dump after the first retries exhausted:
```
13:51:22.977 ←[1;37m[info] server started←[0m
13:51:23.000 ←[0;38m[debug] Connecting DemoConnector to mqtt://esmqdev1.zedisolutions.com:1883←[0m
13:51:23.016 ←[1;33m[warning] lager_error_logger_h dropped 6 messages in the last second that exceeded the limit of 50 messages/sec←[0m
13:51:23.016 ←[0;38m[debug] Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at pid <0.256.0>←[0m
13:51:23.016 ←[0;38m[debug] Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.255.0>←[0m
13:51:23.016 ←[1;37m[info] Application lorawan_server started on node 'lorawan@ZI1L-TS05'←[0m
13:51:23.024 ←[0;38m[debug] Supervisor inet_gethost_native_sup started undefined at pid <0.259.0>←[0m
13:51:23.024 ←[0;38m[debug] Supervisor kernel_safe_sup started inet_gethost_native:start_link() at pid <0.258.0>←[0m
[warning] [Client <0.257.0>] Connection lost for: tcp_closed
13:51:23.884 ←[1;37m[info] gateway 00800000A0000132 {connected,{{192,168,129,181},38853}}←[0m
[warning] [Client <0.257.0>] Connection lost for: tcp_closed
[warning] [Client <0.257.0>] Connection lost for: tcp_closed
[warning] [Client <0.257.0>] Connection lost for: tcp_closed
[warning] [Client <0.257.0>] Connection lost for: tcp_closed
[error] [Client <0.257.0>] reconect error: retries_exhausted
13:52:23.131 ←[1;31m[error] gen_server <0.256.0> terminated with reason: no match of right hand value {ok,<0.266.0>} in lorawan_connector_mqtt:handle_info/2 line 89←[0m
[warning] [Client <0.269.0>] Connection lost for: tcp_closed
13:52:23.143 ←[1;31m[error] CRASH REPORT Process <0.256.0> with 0 neighbours exited with reason: no match of right hand value {ok,<0.266.0>} in lorawan_connector_mqtt:handle_info/2 line 89 in gen_server:terminate/7 line 812←[0m
13:52:23.143 ←[0;38m[debug] Connecting DemoConnector to mqtt://esmqdev1.zedisolutions.com:1883←[0m
13:52:23.144 ←[1;31m[error] Supervisor lorawan_connector_sup had child <<"DemoConnector">> started with lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at <0.256.0> exit with reason no match of right hand value {ok,<0.266.0>} in lorawan_connector_mqtt:handle_info/2 line 89 in context child_terminated←[0m
13:52:23.144 ←[1;31m[error] gen_fsm <0.266.0> in state connected terminated with reason: no match of right hand value {ok,<0.266.0>} in lorawan_connector_mqtt:handle_info/2 line 89←[0m
13:52:23.144 ←[1;31m[error] CRASH REPORT Process <0.266.0> with 1 neighbours exited with reason: no match of right hand value {ok,<0.266.0>} in lorawan_connector_mqtt:handle_info/2 line 89 in gen_fsm:terminate/7 line 559←[0m
13:52:23.145 ←[0;38m[debug] Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at pid <0.268.0>←[0m
13:52:23.145 ←[1;31m[error] gen_server lorawan_connector_factory terminated with reason: no function clause matching lorawan_connector_factory:handle_exit2(<<"DemoConnector">>, {{badmatch,{ok,<0.266.0>}},[{lorawan_connector_mqtt,handle_info,2,[{file,"/root/projects/loraw..."},...]},...]}) line 128←[0m
13:52:23.145 ←[1;31m[error] Cannot connect <<68,101,109,111,67,111,110,110,101,99,116,111,114>>: {error,{already_started,<0.268.0>}}←[0m
13:52:23.145 ←[1;31m[error] CRASH REPORT Process <0.255.0> with 0 neighbours exited with reason: no function clause matching lorawan_connector_factory:handle_exit2(<<"DemoConnector">>, {{badmatch,{ok,<0.266.0>}},[{lorawan_connector_mqtt,handle_info,2,[{file,"/root/projects/loraw..."},...]},...]}) line 128 in gen_server:terminate/7 line 812←[0m
13:52:23.145 ←[1;31m[error] Supervisor lorawan_sup had child connector_factory started with lorawan_connector_factory:start_link() at <0.255.0> exit with reason no function clause matching lorawan_connector_factory:handle_exit2(<<"DemoConnector">>, {{badmatch,{ok,<0.266.0>}},[{lorawan_connector_mqtt,handle_info,2,[{file,"/root/projects/loraw..."},...]},...]}) line 128 in context child_terminated←[0m
13:52:23.145 ←[0;38m[debug] Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.270.0>←[0m
[warning] [Client <0.269.0>] Connection lost for: tcp_closed
```

Crashlog is attached
[crash.txt](https://github.com/gotthardp/lorawan-server/files/998903/crash.txt)

We also have some different logs on the broker as well (though the original wrong version error is still there initially, as expected)
```
May 13, 2017 1:50:35 PM MDT[ServerWorker0] com.ibm.mq.MQXRService.MQTTServerContext
AMQXR0023I: Unsupported MQTT protocol version on channel PlainText, the exception com.ibm.mq.MQXRService.MQTTInvalidVersionException: AMQXR1004E:an invalid message version=4(byte) subVersion=2(byte)was received.  was thrown.
--------------------------------------------------------------------
May 13, 2017 1:50:35 PM MDT[ServerWorker0] com.ibm.mq.MQXRService.MQTTServerContext
AMQXR0021W: Client ''lora1'' at network address ''/192.168.128.128'' disconnected abnormally 
with exception ''An existing connection was forcibly closed by the remote host.''.
```

gotthardp, 2017-05-13:
Thank you. We are coming closer. One more test, please. It might work this time http://swimgate.eu/lorawan-server-0.4.0.tar.gz

DigitalPigeon, 2017-05-13:
Very nice! This works :+1: Thank you very much!

A suggestion might be that instead of waiting until all retries are exhausted to attempt a downgraded 3.1 connection, it might be more ideal to attempt a 3.1.1 connection, then attempt a 3.1 connection. Once both of those fail, count that as one retry.

Currently, it takes ~ 1/2 minute to connect to a 3.1 connection, since all of the 3.1.1 attempts must exhaust first. This is definitely a lower priority request, but if you think it fits within your design intentions, it might be a nice way to implement the fail back.

Regardless, this is great! I really appreciate all of the effort you made here. Wonderful job, and thank you again.

gotthardp, 2017-05-13:
Cool! Thanks a lot for your support.

I am limited by capabilities of the MQTT client I am using, so your suggestion (although very reasonable) is not easily implementable. I will keep this issue open to not forget about this as it will take some time.

gotthardp, 2017-05-20:
The fallback to 3.1 should now be immediate.

DigitalPigeon, 2017-05-23:
This seems to have made things worse.

The connection now never succeeds, and instead this error is repeated continuously in the logs

```
[warning] [Client <0.263.0>] Connection lost for: tcp_closed
08:59:22.952 ←[0;38m[debug] Connector attempt311 disconnected reconnect 5←[0m
08:59:22.952 ←[0;38m[debug] Unknown message: {'EXIT',<0.263.0>,{shutdown,tcp_closed}}←[0m
```

gotthardp, 2017-05-23:
Oh, stupid me. I am very sorry for bothering you.
This should be better: http://swimgate.eu/lorawan-server-0.4.2.tar.gz

DigitalPigeon, 2017-05-24:
Yup, this worked great. Thank you!

closed 2017-05-24 by DigitalPigeon