gotthardp/lorawan-server #90: Can't send downlink over MQTT

felixgalindo, 2017-05-13:
I am trying to publish a downlink message to a connected device and get an error. I am publishing to topic "in/56DDB76F" and the data I am sending is simply "{"data":"hello node"}". I publish this message as soon as I receive a packet from devaddr 56DDB76F. Here are my logs:

14:37:55.183 [info] Application lorawan_server started on node lorawan@raspberrypi
14:37:58.540 [info] gateway 008000000000A3EE {connected,{{127,0,0,1},50527}}
14:38:24.320 [error] gen_server <0.259.0> terminated with reason: bad argument in call to erlang:binary_to_integer(<<"he">>, 16) in lorawan_mac:'-hex_to_binary/1-lbc$^0/2-0-'/2 line 545
14:38:24.366 [error] CRASH REPORT Process <0.259.0> with 0 neighbours exited with reason: bad argument in call to erlang:binary_to_integer(<<"he">>, 16) in lorawan_mac:'-hex_to_binary/1-lbc$^0/2-0-'/2 line 545
14:38:24.366 [debug] Connecting MQTT to mqtt://localhost:1883
14:38:24.369 [error] Supervisor lorawan_connector_sup had child <<"MQTT">> started with lorawan_connector_mqtt:start_link({mqtt,[],"localhost",1883,"/",[]}, {connector,<<"MQTT">>,true,<<"mqtt://localhost:1883">>,<<"out/{devaddr}">>,<<"in/#">>,<<"in/{devad...">>,...}) at <0.259.0> exit with reason bad argument in call to erlang:binary_to_integer(<<"he">>, 16) in lorawan_mac:'-hex_to_binary/1-lbc$^0/2-0-'/2 line 545 in context child_terminated
14:38:24.370 [error] gen_fsm <0.260.0> in state connected terminated with reason: bad argument in call to erlang:binary_to_integer(<<"he">>, 16) in lorawan_mac:'-hex_to_binary/1-lbc$^0/2-0-'/2 line 545
14:38:24.370 [error] CRASH REPORT Process <0.260.0> with 1 neighbours exited with reason: bad argument in call to erlang:binary_to_integer(<<"he">>, 16) in lorawan_mac:'-hex_to_binary/1-lbc$^0/2-0-'/2 line 545
14:38:24.371 [debug] Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"localhost",1883,"/",[]}, {connector,<<"MQTT">>,true,<<"mqtt://localhost:1883">>,<<"out/{devaddr}">>,<<"in/#">>,<<"in/{devad...">>,...}) at pid <0.265.0>
14:38:24.372 [error] gen_server lorawan_connector_factory terminated with reason: no function clause matching lorawan_connector_factory:handle_exit2(<<"MQTT">>, {badarg,[{erlang,binary_to_integer,[<<"he">>,16],[]},{lorawan_mac,'-hex_to_binary/1-lbc$^0/2-0-',...},...]}) line 128
14:38:24.373 [error] CRASH REPORT Process <0.258.0> with 0 neighbours exited with reason: no function clause matching lorawan_connector_factory:handle_exit2(<<"MQTT">>, {badarg,[{erlang,binary_to_integer,[<<"he">>,16],[]},{lorawan_mac,'-hex_to_binary/1-lbc$^0/2-0-',...},...]}) line 128 in gen_server:terminate/7 line 812
14:38:24.373 [error] Cannot connect <<77,81,84,84>>: {error,{already_started,<0.265.0>}}
14:38:24.374 [error] Supervisor lorawan_sup had child connector_factory started with lorawan_connector_factory:start_link() at <0.258.0> exit with reason no function clause matching lorawan_connector_factory:handle_exit2(<<"MQTT">>, {badarg,[{erlang,binary_to_integer,[<<"he">>,16],[]},{lorawan_mac,'-hex_to_binary/1-lbc$^0/2-0-',...},...]}) line 128 in context child_terminated
14:38:24.374 [debug] Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.267.0>

gotthardp, 2017-05-13:
The data should be a hexastring-- "hello node" should be "68656c6c6f206e6f6465".

felixgalindo, 2017-05-13:
Got it thanks.

closed 2017-05-13 by felixgalindo