gotthardp/lorawan-server #91: OTAA is stable with stable branch

batulzii, 2017-05-13:
I got OTAA not reliable after upgraded current branch. I compile Stable branch, OTAA back very stable.
What is main changes in OTAA timing between these branches?

gotthardp, 2017-05-13:
What troubles are you experiencing?

batulzii, 2017-05-13:
Join request / ACK and  Confirmed Data/ ACK  seen from packet forwarder in both branch. Motes get ACKs in STABLE branch, can not get Current branch. Cycling to request to join in current branch. So recover stale branch now. No problem with join and confirmed data

gotthardp, 2017-05-13:
"Cycling to request to join"??

batulzii, 2017-05-13:
sorry for bad explain. Device sending JOIN REQUEST all the times because it did not get JOIN ACCEPT.

gotthardp, 2017-05-13:
I am aware of an OTAA related issue, but I have troubles debugging it because on my devices it works fine. Would you be willing to help with debugging, please?

batulzii, 2017-05-14:
Please, send me a little deep explain about your OTAA algorithm and what was made latest change on it. I will debug it.

gotthardp, 2017-05-14:
Not much has changed:
 * The biggest OTAA change is the RX1DROffset field in Join Accept. It used to be always 0. Now its content depends on the device configuration, so it can be 0 or anything else. The latest `master` even prints the values put in Join Accept, so you can double check if its 0 or something else.
 * Then the order of fields in the `txframe` message sent from server to gateway may be different
 * And I added some checks for the txpower, so the TX power may be different (but the same for all downlinks)

The vast majority of changes to master is documentation:
https://github.com/gotthardp/lorawan-server/compare/stable...master

batulzii, 2017-05-14:
Here are 2 different devices, One got ACCEPT in RX1, other is not.
It seems to be our device had problem with RX2 :
2017-05-14 11:05:45.650 [info] <0.749.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,3,26>> -> <<172,63,146,181>>
2017-05-14 11:05:45.650 [debug] <0.749.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,0,869.525}
2017-05-14 11:06:05.365 [info] <0.751.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 11:06:05.365 [debug] <0.751.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 11:08:13.924 [info] <0.792.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 11:08:13.924 [debug] <0.792.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 11:10:22.513 [info] <0.807.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 11:10:22.513 [debug] <0.807.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 11:12:31.009 [info] <0.809.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 11:12:31.009 [debug] <0.809.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 11:14:39.574 [info] <0.839.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 11:14:39.574 [debug] <0.839.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}

gotthardp, 2017-05-14:
There should be a log before this when the second device got join-accept in RX1. The RX2 is used only after sending join accept in RX1, which was unsuccessful.

batulzii, 2017-05-14:
I reset server and joining again 2 devices and now used only RX2 and :
2017-05-14 19:26:49.426 [debug] <0.256.0> Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.261.0>
2017-05-14 19:26:49.426 [info] <0.31.0> Application lorawan_server started on node 'lorawan@ulzii-PC'
2017-05-14 19:26:58.150 [info] <0.258.0> gateway 0023CDFFFF000E7E {connected,{{192,168,0,100},49859}}
2017-05-14 19:27:56.031 [info] <0.262.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,3,26>> -> <<172,63,146,181>>
2017-05-14 19:27:56.031 [debug] <0.262.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 19:30:07.980 [info] <0.289.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,3,26>> -> <<172,63,146,181>>
2017-05-14 19:30:07.980 [debug] <0.289.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 19:30:37.337 [info] <0.290.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 19:30:37.337 [debug] <0.290.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 19:32:20.635 [info] <0.292.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,3,26>> -> <<172,63,146,181>>
2017-05-14 19:32:20.635 [debug] <0.292.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 19:32:45.981 [info] <0.293.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 19:32:45.990 [debug] <0.293.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 19:34:33.328 [info] <0.308.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,3,26>> -> <<172,63,146,181>>
2017-05-14 19:34:33.329 [debug] <0.308.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 19:34:54.543 [info] <0.310.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,36,151,66>>
2017-05-14 19:34:54.543 [debug] <0.310.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}

gotthardp, 2017-05-14:
This is definitely wrong. Could you please `git pull` and `make upgrade` and test my latest fix?

batulzii, 2017-05-14:
i got it just an hour before.

gotthardp, 2017-05-14:
Yes, I have. My local time is distorted.

batulzii, 2017-05-14:
seen changes, but have not reliable OTAA in timiing. please,check bellow:
2017-05-14 21:19:57.098 [info] <0.425.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:19:57.163 [debug] <0.425.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,0,869.525}
2017-05-14 21:20:04.604 [info] <0.426.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,3,26>> -> <<172,212,126,211>>
2017-05-14 21:20:04.604 [debug] <0.426.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,0,869.525}
2017-05-14 21:22:05.042 [info] <0.450.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:22:05.042 [debug] <0.450.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 21:24:13.665 [info] <0.512.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:24:13.665 [debug] <0.512.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,0,869.525}
2017-05-14 21:26:22.288 [info] <0.533.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:26:22.288 [debug] <0.533.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 21:28:30.963 [info] <0.535.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:28:30.963 [debug] <0.535.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,0,869.525}
2017-05-14 21:30:39.657 [info] <0.552.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:30:39.657 [debug] <0.552.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 21:31:27.131 [debug] <0.553.0>@lorawan_mac:handle_uplink:338 ADR indicator set to 1
2017-05-14 21:31:27.201 [debug] <0.553.0>@lorawan_mac_commands:build_fopts:32 <<172,212,126,211>> <- [dev_status_req]
2017-05-14 21:31:27.211 [debug] <0.55.0> Supervisor kernel_safe_sup started pg2:start_link() at pid <0.554.0>
2017-05-14 21:32:48.310 [info] <0.571.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:32:48.310 [debug] <0.571.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,0,869.525}
2017-05-14 21:34:57.023 [info] <0.586.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:34:57.023 [debug] <0.586.0>@lorawan_mac:handle_join:188 Join-Accept in RX2: {0,0,869.525}
2017-05-14 21:37:01.499 [debug] <0.588.0>@lorawan_mac_commands:handle:17 <<172,212,126,211>> -> [{dev_status_ans,39,25}]
2017-05-14 21:37:01.499 [debug] <0.588.0>@lorawan_mac_commands:handle_status:129 DevStatus: battery 39, margin: -7
2017-05-14 21:37:05.751 [info] <0.589.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<17,34,51,68,85,102,119,17>> <<190,122,0,0,0,0,0,96>> -> <<172,34,33,155>>
2017-05-14 21:37:05.751 [debug] <0.589.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,0,869.525}
2017-05-14 21:37:13.582 [debug] <0.590.0>@lorawan_mac:handle_uplink:338 ADR indicator set to 1
2017-05-14 21:37:13.582 [debug] <0.590.0>@lorawan_mac_commands:build_fopts:32 <<172,34,33,155>> <- [dev_status_req]

gotthardp, 2017-05-14:
Well, and could it be a device or a gateway (configuration) issue? Could you please check on the device side f the device didn't receive the Join-Accept because it was listening on wrong channel, or received it and didn't like it.

batulzii, 2017-05-14:
Thank you for your time. I will check and let you know

batulzii, 2017-05-14:
I have fix OTAA on LoraMac firmware. adjusting RX/offset. 2 different PCB have a little different adjustment.
Please, Let me know if any advice other than this way.

closed 2017-05-14 by batulzii