gotthardp/lorawan-server #92: Add {devEUI} as a wildcard for connectors

DigitalPigeon, 2017-05-13:
This is a feature request.

Would be very useful if the connector supported **{devEUI}** as a connector wildcard, in the same way that it supports **{devaddr}**.

In our real world deployment, we sell our customer a batch of devices. These are not activated at time of sale, but are initialized in the application server using devEUI. Once installed by the customer, the devices use OTAA, and so the {devaddr} is meaningless to the application server.

Network server is now pushing messages to Application Server via MQTT, but these messages are identified via {devaddr}, which the Application Server doesn't know about. If {devEUI} was supported, then the LoRa mote would be identified by its MQTT topic name, and the application server could subscribe and publish for each mote based on topic name, much the same way that the Network Server is doing.

This would mean, specifically for outgoing messages, that the network server would have to read the topic name and parse out {devEUI} like it does for {devaddr} today, but would then need to internally look up the current {devaddr} for that {devEUI} to properly route the message.

I hope I described this well enough.

Thanks,
Chris

DigitalPigeon, 2017-05-13:
I guess I would comment to requesting this be expanded to the data contained inside of the JSON format data as well (including devEUI as one of the JSON fields).

gotthardp, 2017-05-13:
Sounds reasonable. I will implement it tomorrow (both the connector pattern as well as JSON), but it needs some refactoring, so it's not as easy as I originally thought.

gotthardp, 2017-05-14:
Done. Please try http://swimgate.eu/lorawan-server-0.4.1.tar.gz

DigitalPigeon, 2017-05-15:
This works beautifully! Thank you very much.

gotthardp, 2017-05-15:
Cool! Now I need to fix the crashes you reported earlier ;-)

closed 2017-05-15 by gotthardp