gotthardp/lorawan-server #93: lorawan-daemon crashing

Sviests, 2017-05-15:
I am unable to launch the lorawan-daemon. I`ve been trying to add the lorawan-server.service  which uses the daemon and I`m always getting this crash. This happens when I launch the daemon manually also.

{{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,"application_controller.erl"},{line,1809}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]},{application_controller,check_conf,0,[{file,"application_controller.erl"},{line,1808}]},{application_controller,init,2,[{file,"application_controller.erl"},{line,485}]}]}}},[{application_controller,init,2,[{file,"application_controller.erl"},{line,485}]}]}
{"could not start kernel pid",application_controller,"{{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,\"application_controller.erl\"},{line,1809}]},{lists,foldl,3,[{file,\"lists.erl\"},{line,1263}]},{application_controller,check_conf,0,[{file,\"application_controller.erl\"},{line,1808}]},{application_controller,init,2,[{file,\"application_controller.erl\"},{line,485}]}]}}},[{application_controller,init,2,[{file,\"application_controller.erl\"},{line,485}]}]}"}
could not start kernel pid (application_controller) ({{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,"application_controller.erl"},{line,1809}]},{

gotthardp, 2017-05-15:
What erlang version do you have?

Sviests, 2017-05-15:
I have erlang version 19, which is needed for the current version of cowlib to build.

gotthardp, 2017-05-15:
And you removed any traces of old releases? The thing is this appears to be an installation issue. The code crashing is not mine.

Sviests, 2017-05-15:
I did have a problem with removing the old erlang packages. I will check again if I have removed everything properly. Thanks for the tip!

Sviests, 2017-05-15:
After a full reinstall of all erlang packages it worked. Thanks for the help!

gotthardp, 2017-05-15:
Cool. I am glad it works!

closed 2017-05-15 by gotthardp