gotthardp/lorawan-server #94: Multiple nodes using MQTT connectors

Sviests, 2017-05-15:
I finally got a test node to measure the range of my LoraWAN network, but I cannot seem to get the test node to post through the MQTT connector that I have set up.

My current setup:
A sensor connected to my LoraWAN network which has joined over OTAA, an MQTT connector setup for it and a handler using the connector. This works mostly as I expect, I receive the uplink messages on the MQTT topic I have specified (still can`t get the downlinks to work tho...) and I am using it as the information source for the integration I have written for a monitoring software I am using.

A test node, added over ABP, has its separate connector using MQTT to post to its respective topic and a separate handler.

Both setups are pretty much identical, they both use the same MQTT URI, just post to different topics. The sensor node works and I see its posted JSON, but the test node does not work at all and I am receiving the following errors in the debug.log file:

2017-05-15 16:01:24.290 [debug] <0.2641.0>@lorawan_connector_mqtt:init:22 Connecting MQTT Test Node to mqtt://mqtt
2017-05-15 16:01:24.290 [error] <0.460.0> CRASH REPORT Process <0.460.0> with 1 neighbours exited with reason: no function clause matching lorawan_admin:parse(undefined) line 31 in gen_fsm:terminate/7 line 559
2017-05-15 16:01:24.290 [error] <0.295.0> Supervisor lorawan_connector_sup had child <<"MQTT Test Node">> started with lorawan_connector_mqtt:start_link({mqtt,[],"mqtt",1883,"/",[]}, {connector,<<"MQTT Test Node">>,true,<<"mqtt://mqtt">>,<<"test_node/out/{devaddr}">>,...}) at <0.459.0> exit with reason no function clause matching lorawan_admin:parse(undefined) line 31 in context child_terminated
2017-05-15 16:01:24.292 [debug] <0.295.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"mqtt",1883,"/",[]}, {connector,<<"MQTT Test Node">>,true,<<"mqtt://mqtt">>,<<"test_node/out/{devaddr}">>,...}) at pid <0.2641.0>
2017-05-15 16:03:29.347 [debug] <0.2753.0>@lorawan_mac:check_link_fcnt:245 <<3,255,0,1>> fcnt reset
2017-05-15 16:03:29.348 [debug] <0.2753.0>@lorawan_mac:handle_uplink:341 DataRate <<3,255,0,1>> switched to dr 5
2017-05-15 16:03:29.348 [debug] <0.2753.0>@lorawan_mac_commands:build_fopts:32 <<3,255,0,1>> <- [dev_status_req]
2017-05-15 16:03:29.348 [error] <0.1262.0>@lorawan_connector_factory:init_connector:111 Cannot connect <<77,81,84,84,32,87,97,116,101,114,32,77,101,116,101,114>>: {error,{already_started,<0.1413.0>}}
2017-05-15 16:03:29.348 [error] <0.2753.0>@lorawan_connector_factory:publish:24 Error {already_started,<0.1413.0>}
2017-05-15 16:03:40.486 [debug] <0.2754.0>@lorawan_mac_commands:handle:17 <<3,255,0,1>> -> [{dev_status_ans,91,63}]
2017-05-15 16:03:40.486 [debug] <0.2754.0>@lorawan_mac_commands:handle_status:129 DevStatus: battery 91, margin: 31
2017-05-15 16:03:40.487 [error] <0.1262.0>@lorawan_connector_factory:init_connector:111 Cannot connect <<77,81,84,84,32,87,97,116,101,114,32,77,101,116,101,114>>: {error,{already_started,<0.1413.0>}}
2017-05-15 16:03:40.487 [error] <0.2754.0>@lorawan_connector_factory:publish:24 Error {already_started,<0.1413.0>}
2017-05-15 16:03:57.311 [error] <0.1262.0>@lorawan_connector_factory:init_connector:111 Cannot connect <<77,81,84,84,32,87,97,116,101,114,32,77,101,116,101,114>>: {error,{already_started,<0.1413.0>}}

Sviests, 2017-05-15:
Never mind, seems like a simple restart of the server solved this...

gotthardp, 2017-05-15:
Thanks for reporting this. It should not happen. I will try to avoid this in the future.

gotthardp, 2017-05-20:
I reworked this a bit. The nightly build http://swimgate.eu/lorawan-server-0.4.1.tar.gz should behave better.

gotthardp, 2017-05-24:
This should be fixed in 0.4.2.

closed 2017-05-24 by gotthardp