gotthardp/lorawan-server #95: Scrip to decrypt Hex frames on the server

kazi93, 2017-05-15:
Hi currently i am decrypting the Hex frame by using the following:
http://string-functions.com/hex-string.aspx

For example: I am receiving 6869 in the Loraserver which is equivalent to "Hi". Is there any script that will convert multiple hex readings (from multiple nodes) and post it on the visualisation tool ( for example feeboard.io or initialstate or kibana or thingspeak ? ) ?

gotthardp, 2017-05-15:
Such conversion is the purpose of Handlers. If you choose one (preferably open source and visually pleasing) I can help you with the integration.

gotthardp, 2017-05-20:
Do you require further assistance on this?

kazi93, 2017-06-13:
@gotthardp I have decided to go with ThingSpeak as we can use "MATLAB Analysis and Visualisation". Your guidance will be much appreciated. Thank you :)

Sensor node -> LoRaWAN gateway -> Lorawan-server -> ThingSpeak ?

gotthardp, 2017-07-02:
Hello. It took some time, but I finally made something that *may* work. Please, try to follow the https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md#mathworks-thingspeak guide, test and let me know the issues.

kazi93, 2017-07-27:
@gotthardp Could you please take a look at the below settings on thingspeak and the loraserver. I must made some terrible mistake. I can see nothing on the visualisation tool.

[image]
[image]
[image]

kazi93, 2017-07-27:
@gotthardp I guess, I should explain a little bit more. Sensor with group name "arduino" configured with ABP method and the esm5k is configured with OTAA method. Sensor arduino only sending "Hi" and sensor esm5k is sending temperature value. Please guide me to config in thingspeak.com.  Let me know if you need more information. Thanks

gotthardp, 2017-07-28:
Could you please let me know what is the server doing? Is there something in the logs? Is the server sending some HTTP request (you can check this via tcpdump or tshark)?

kazi93, 2017-07-28:
Hi @gotthardp I have attached the pcap file. I am also trying to understand the file. Thank you
[file.pcap.zip](https://github.com/gotthardp/lorawan-server/files/1183007/file.pcap.zip)

kazi93, 2017-07-28:
Little more info: 
root@vps433520:~# docker container ls
CONTAINER ID        IMAGE                             COMMAND                  CREATED             STATUS              PORTS                                            NAMES
6eb9acad6429        gotthardp/lorawan-server:latest   "/bin/sh -c bin/lo..."   2 weeks ago         Up 2 weeks          0.0.0.0:1680->1680/udp, 0.0.0.0:8080->8080/tcp   lorawan

kazi93, 2017-07-28:
However, I cannot find "Mnesia database and server logs" which I defined as /opt during configuration :(

gotthardp, 2017-07-29:
Well, the .pcap file is empty. There is not much to understand about it. How did you create it?

kazi93, 2017-07-29:
I captured it from the ubuntu terminal however the loraserver is running inside the docker. The server is receiving frames from 3 nodes and hosted in the VPS cloud. Is that a problem?

gotthardp, 2017-07-29:
This doesn't seem to be a problem. I suspect you set a wrong interface for `sudo tcpdump -i eth0 -w file.pcap`. Shouldn't you use something else than eth0?

kazi93, 2017-07-29:
@gotthardp Yes, you were right. As it is installed in docker therefore I have captured both docker0 and esn3 interface. Thanks again.
[docker0.pcap.zip](https://github.com/gotthardp/lorawan-server/files/1184955/docker0.pcap.zip)
[ens3.pcap.zip](https://github.com/gotthardp/lorawan-server/files/1184956/ens3.pcap.zip)

gotthardp, 2017-07-29:
No MQTT communication in there, so we will need to find the server log. What was the `docker run` command you used to start the server?

kazi93, 2017-07-29:
@gotthardp 
$ docker run lorawan
I have used the below documentation to install docker.

@https://www.digitalocean.com/community/tutorials/how-to-install-and-use-docker-on-ubuntu-16-04

gotthardp, 2017-07-29:
I didn't find anything wrong in the configs you posted, but I see an error in how you run the docker: the `/path/to/local` shall point to your directory, so if you want to have logs in `/opt/lorawan` do `--volume /opt/lorawan:/storage`. The `/storage` has to be untouched.

kazi93, 2017-07-29:
@gotthardp Do I need to reinstall it? or I can run the modified command again? Please suggest bit briefly. Thank you

gotthardp, 2017-07-29:
I am not a docker expert, but I believe you have to stop and remove the old image and then start it using the new command. Unfortunately all your config in that image will be lost.

kazi93, 2017-07-29:
@gotthardp reinstall the loraserver, receiving frames in the server too. Now, I have the log and Mnesia database too in the desired local folder. Thank you. Please take a look at the attached *pcap file. Please take a look at my arduino handler, am I doing some missconfig ?? I am only expecting a string from the arduino which is "Hi"
[Archive.zip](https://github.com/gotthardp/lorawan-server/files/1185138/Archive.zip)

gotthardp, 2017-07-29:
Could you please share also the debug.log?

kazi93, 2017-07-29:
@gotthardp Yes. Thank you
[Archive 2.zip](https://github.com/gotthardp/lorawan-server/files/1185156/Archive.2.zip)

gotthardp, 2017-07-29:
The data you send don't match the Parse Uplink function you specified: the function expects the first byte in the application data indicates length of the frame. If this is not true you should modify the function, for example to:
```erlang
fun(_Port, Data) ->
  #{data => binary_to_list(Data)}
end.
```

kazi93, 2017-07-29:
@gotthardp Thank you. I am just expecting "Hi" from the lorawan arduino node. Please have a look and suggest accordingly. I have already changed the above code, not helpful yet. 
[image]
[image]
[image]
[image]
[image]
[image]

gotthardp, 2017-07-29:
Please post the PCAP and server logs too. :-)

kazi93, 2017-07-29:
@gotthardp I have been trying it since yesterday. It won't work! I must have been doing something wrong. As you can see I am just sending a string "Hi", I would like to print it in thingspeak. Thank you. Another sensor is also sending temperature and humidity data, if this is a success that supposed to work too. Thanks again for your kind support.  :)

gotthardp, 2017-07-29:
Don't give up :) The Received Frames page shows "hello" as payload, so we are almost there. We fixed the parsing function, but there is apparently (at least) one more issue. The PCAP and debug.log will help finding the problem.

kazi93, 2017-07-29:
@gotthardp Thank you. Yes, payload is "hello". :) I have attached both interface file docker0 and ens3 both. 
[debug.log.zip](https://github.com/gotthardp/lorawan-server/files/1185196/debug.log.zip)
[docker2.pcap.zip](https://github.com/gotthardp/lorawan-server/files/1185197/docker2.pcap.zip)
[ens4.pcap.zip](https://github.com/gotthardp/lorawan-server/files/1185198/ens4.pcap.zip)

gotthardp, 2017-07-29:
Thanks. Now the ThingSpeak is refusing the connection. Could you please try switching to plain "mqtt"?

kazi93, 2017-07-29:
@gotthardp Now switched to "mqtt://mqtt.thingspeak.com", however, no output. What should be in the thingspeak channel settings >>field 1, field 2, ....?
[ens5.pcap.zip](https://github.com/gotthardp/lorawan-server/files/1185236/ens5.pcap.zip)

gotthardp, 2017-07-29:
I can see the MQTT Publish now. It contains "data=hello", so I recommend adding a field1 "data" and deleting every other fields (for now).

kazi93, 2017-07-29:
@gotthardp I can see little update. Entries : 72 :) however, cannot see "Hello" yet. :) Changed the Field 1 : >> "data". Any suggestion? Almost there :P
[image]

gotthardp, 2017-07-29:
Are you sure that ThingSpeak handles also text data? The displayed chart with x-y axes looks like a numerical value was expect.

kazi93, 2017-07-29:
@gotthardp I will do some research on this. However, I have another sensor that is OTTA configured and sending temp and humid as you can see below. It publish numerical values. :). Let us try. Many thanks for your kind support. I have changed the Field 1 >> Data too. But output is also null. :(
[image]
[image]

gotthardp, 2017-07-29:
But you need to adapt also the Parse Uplink function so it sends some numerical data. You can follow the [Instructions](https://github.com/gotthardp/lorawan-server/blob/master/doc/Backends.md#parse-uplink) or describe the format here and I will suggest the parsing function.

kazi93, 2017-07-29:
@gotthardp Thanks again for your kind support. I just went through the sensor documentation. It says something like below: Please help, cannot wait to see it is working with thingspeak :)

[image]
[image]
[image]

gotthardp, 2017-07-29:
But how does `21BF120213` match the above description? Could you please help me decoding this?

kazi93, 2017-07-29:
@gotthardp I will try to decode it but I am not an expert :( I think we only need to use the parsing function that should do the job. :) Btw. I have sent couple of email in your "@centrum.cz" address. Please inbox me. Many thanks :)

gotthardp, 2017-07-29:
I can suggest the Erlang function for Uplink Parsing, but you need to figure out what means each byte in the message. Just read the device documentation and tell me the meaning of each byte in `21BF120213`, which is the message I see in Received Frames.

kazi93, 2017-07-29:
I think it is written above shared documentation. Length is also defined. So, GPS and TEMP data.
[image]

gotthardp, 2017-07-30:
But the data I see you got are 3 or 5 bytes long. I don't see how the documentation you shared is matching the frames you are receiving.

kazi93, 2017-07-31:
@gotthardp I have attached the payload decoding code. How could I use it to have sensor readings on thingspeak? I think script says everything. I am receiving data in the "Received Frames" :)
[image]
[payload.txt.zip](https://github.com/gotthardp/lorawan-server/files/1187808/payload.txt.zip)

gotthardp, 2017-07-31:
I see. For the data you are getting you can use the following function:
```erlang
fun(_Port, <<16#01, Temp:16/signed-integer, 16#07, VDD:16>>) ->
  #{temp => Temp, vdd => VDD}
end.
```
I hope it is pretty straightforward.

kazi93, 2017-08-01:
@gotthardp Nothing happened. Even thought it is receiving entries in thingspeak. definitely missing something. 
[image]
[image]
[image]

gotthardp, 2017-08-01:
Are the names in thingspeak case sensitive? In the message they are all lowercase.

kazi93, 2017-08-01:
@gotthardp I do not think so. I guess, it is a decoding problem. :(
[image]

gotthardp, 2017-08-04:
Hi there! The trick is to use "fieldX" names, so the correct Parse Uplink function is:

```erlang
fun(_Port, <<16#01, Temp:16/signed-integer, 16#07, VDD:16>>) ->
  #{field1 => Temp, field2 => VDD}
end.
```

closed 2017-08-10 by gotthardp

kazi93, 2017-08-24:
@gotthardp I have gone through the Backends.md, and I have decided to learn erlang programming. However, now I only have a analog gas sensor which sending a decimal value of e.g 40 or 340 (depending on the density of gas). Below is the received frames in ASCII format and the sketch I have used. Please correct me if my parsing is wrong !!
[image]

fun(_Port, <<16#01, message:16/signed-integer>>) ->
  #{message => message}
end.

kazi93, 2017-08-24:
@using the same above code : DS18B20 TEMP sensor. Parsing is wrong as the thingspeak output is not the same as the serial output.
Parsing:
**fun(_Port, <<Temp, B>>) ->
  #{field1 => Temp}
end.**

[image]

gotthardp, 2017-08-24:
The parsing fun has to match the data. If you have a 2-byte value like 000C you can do
```erlang
fun(_Port, <<Val:16/signed-integer>>) ->
  #{field1 => Val}
end.
```

kazi93, 2017-08-24:
@gotthardp I did it however, I think something is wrong with the my sketch as the serial output is just fine but the server ASCII is not, it is always the same 000C. :(

Note I have just modified SKETCH : @https://github.com/gonzalocasas/arduino-uno-dragino-lorawan/blob/master/send-sensor-data/send-sensor-data.ino and add the ds18b20 code.

What do you think about the analog gas sensor (A0) which also sending the decimal value to the server the parsing is okay? I cannot have any data in the thingspeak too.

fun(_Port, <<16#01, message:16/signed-integer>>) ->
#{message => message}
end.

gotthardp, 2017-08-24:
The variables have to start with a capital letter, so it has to be:
```erlang
fun(_Port, <<Message:16/signed-integer>>) ->
    #{field1 => Message}
end.
```

kazi93, 2017-08-29:
@gotthardp Many thanks. I have managed to get temp. from the ds18b20 sensor. Could you please help me to parsing it. I have tried and it did not work unfortunately due to my poor parsing knowledge. I have attached the sketch and rxframes. Thanks again. :)

[image]

[sketch.txt.zip](https://github.com/gotthardp/lorawan-server/files/1259103/sketch.txt.zip)

gotthardp, 2017-08-29:
You should really learn more about Erlang binaries. It is not that hard.

As a first step you can try. It is just parsing 3-byte singned integer value. I have no clue if this is correct as I had no time to review the sketch.
```erlang
fun(_Port, <<Message:24/signed-integer>>) ->
    #{field1 => Message}
end.
```