gotthardp/lorawan-server #96: Crash on MQTT down, does not successfully recover connection to MQTT

DigitalPigeon, 2017-05-15:
We had a reboot of our MQTT server, and lorawan-server did not recover. 

It generated a crash report when the MQTT server went down.
```
09:51:10.816 ←[1;31m[error] gen_server <0.480.0> terminated with reason: {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}} in gen_fsm:sync_send_event/3 line 231←[0m
09:51:10.834 ←[1;31m[error] CRASH REPORT Process <0.480.0> with 0 neighbours exited with reason: {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}} in gen_server:terminate/7 line 812←[0m
09:51:10.834 ←[0;38m[debug] Connecting DemoConnector to mqtt://esmqdev1.zedisolutions.com:1883←[0m
09:51:10.834 ←[1;31m[error] Supervisor lorawan_connector_sup had child <<"DemoConnector">> started with lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at <0.480.0> exit with reason {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}} in context child_terminated←[0m
09:51:10.834 ←[1;31m[error] gen_fsm <0.490.0> in state connected terminated with reason: {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}}←[0m
09:51:10.835 ←[1;31m[error] CRASH REPORT Process <0.490.0> with 1 neighbours exited with reason: {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}} in gen_fsm:terminate/7 line 559←[0m
09:51:10.835 ←[0;38m[debug] Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at pid <0.16702.0>←[0m
09:51:10.835 ←[1;31m[error] gen_server lorawan_connector_factory terminated with reason: no function clause matching lorawan_connector_factory:handle_exit2(<<"DemoConnector">>, {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}}) line 128←[0m
09:51:10.835 ←[1;31m[error] Cannot connect <<68,101,109,111,67,111,110,110,101,99,116,111,114>>: {error,{already_started,<0.16702.0>}}←[0m
09:51:10.835 ←[1;31m[error] CRASH REPORT Process <0.257.0> with 0 neighbours exited with reason: no function clause matching lorawan_connector_factory:handle_exit2(<<"DemoConnector">>, {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}}) line 128 in gen_server:terminate/7 line 812←[0m
09:51:10.835 ←[1;31m[error] Supervisor lorawan_sup had child connector_factory started with lorawan_connector_factory:start_link() at <0.257.0> exit with reason no function clause matching lorawan_connector_factory:handle_exit2(<<"DemoConnector">>, {timeout,{gen_fsm,sync_send_event,[<0.490.0>,{<0.480.0>,ping},10000]}}) line 128 in context child_terminated←[0m
09:51:10.835 ←[0;38m[debug] Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.16704.0>←[0m
[warning] [Client <0.16703.0>] Connection lost for: tcp_closed
[warning] [Client <0.16703.0>] Connection lost for: tcp_closed
```

Then it generated a crash report when the MQTT server came back up.
```
10:08:04.983 ←[0;38m[debug] Connecting DemoConnector to mqtt://esmqdev1.zedisolutions.com:1883←[0m
10:08:04.983 ←[1;31m[error] CRASH REPORT Process <0.16734.0> with 0 neighbours exited with reason: {timeout,{gen_fsm,sync_send_event,[<0.16744.0>,{<0.16734.0>,ping},10000]}} in gen_server:terminate/7 line 812←[0m
10:08:04.983 ←[1;31m[error] gen_fsm <0.16744.0> in state connected terminated with reason: {timeout,{gen_fsm,sync_send_event,[<0.16744.0>,{<0.16734.0>,ping},10000]}}←[0m
10:08:04.983 ←[1;31m[error] CRASH REPORT Process <0.16744.0> with 1 neighbours exited with reason: {timeout,{gen_fsm,sync_send_event,[<0.16744.0>,{<0.16734.0>,ping},10000]}} in gen_fsm:terminate/7 line 559←[0m
10:08:04.983 ←[1;31m[error] Supervisor lorawan_connector_sup had child <<"DemoConnector">> started with lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at <0.16734.0> exit with reason {timeout,{gen_fsm,sync_send_event,[<0.16744.0>,{<0.16734.0>,ping},10000]}} in context child_terminated←[0m
10:08:04.984 ←[0;38m[debug] Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at pid <0.16848.0>←[0m
[warning] [Client <0.16849.0>] Connection lost for: tcp_closed
[warning] [Client <0.16849.0>] Connection lost for: tcp_closed
[error] [Client <0.16849.0>] reconect error: retries_exhausted
```

After that, it threw the following error each time new data was received

```
10:12:08.192 ←[1;31m[error] Cannot connect <<68,101,109,111,67,111,110,110,101,99,116,111,114>>: {error,{already_started,<0.16848.0>}}←[0m
10:12:08.192 ←[1;31m[error] Error {already_started,<0.16848.0>}←[0m
```

I've included the applicable periods of the crash dump and debug log if it helps.

[debug.txt](https://github.com/gotthardp/lorawan-server/files/1001714/debug.txt)
[crash.txt](https://github.com/gotthardp/lorawan-server/files/1001713/crash.txt)

Hopefully this wasn't introduced in the work you did for us in #89 for 3.1 fallback.

gotthardp, 2017-05-17:
Thanks for reporting this. If you got some time, could you try upgrading to http://swimgate.eu/lorawan-server-0.4.1.tar.gz, please?

DigitalPigeon, 2017-05-18:
I've upgraded today, but have not had a chance to exercise MQTT failures yet. I will make sure to test more thoroughly tomorrow.

altishchenko, 2017-05-18:
I am sorry to barge in, but it can be the case of the ‘emqttc’ automatic reconnect feature. The protocol driver for the emqttc is implemented as gen_fsm and that is what I see from the log: gen_fsm <0.16744.0> in state connected terminated with reason: {timeout,{gen_fsm,sync_send_event,[<0.16744.0>,{<0.16734.0>,ping},10000]}}.
Possibly, disabling automatic reconnect by emqttc and leaving it to lorawan_sup may solve the restart problem. Petr?

Regards,
Alexander Tishchenko
mob: +7 (916) 704-61-57
atishch@gmail.com

gotthardp, 2017-05-19:
The way the restart mechanism in mqttc works is not very nice. I tried to use it, but it causes more troubles than benefits, so I really consider disabling it.

DigitalPigeon, 2017-05-19:
I don't believe it is fixed in the current build.

According to the logs, a couple of days ago we lost connection to mqtt
```
2017-05-17 23:36:01.858 [error] <0.258.0> gen_server <0.258.0> terminated with reason: {timeout,{gen_fsm,sync_send_event,[<0.267.0>,{<0.258.0>,ping},10000]}} in gen_fsm:sync_send_event/3 line 231
2017-05-17 23:36:01.881 [error] <0.258.0> CRASH REPORT Process <0.258.0> with 0 neighbours exited with reason: {timeout,{gen_fsm,sync_send_event,[<0.267.0>,{<0.258.0>,ping},10000]}} in gen_server:terminate/7 line 812
2017-05-17 23:36:01.881 [debug] <0.2119.0>@lorawan_connector_mqtt:init:22 Connecting DemoConnector to mqtt://esmqdev1.zedisolutions.com:1883
2017-05-17 23:36:01.882 [error] <0.256.0> Supervisor lorawan_connector_sup had child <<"DemoConnector">> started with lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at <0.258.0> exit with reason {timeout,{gen_fsm,sync_send_event,[<0.267.0>,{<0.258.0>,ping},10000]}} in context child_terminated
2017-05-17 23:36:01.882 [error] <0.267.0> gen_fsm <0.267.0> in state connected terminated with reason: {timeout,{gen_fsm,sync_send_event,[<0.267.0>,{<0.258.0>,ping},10000]}}
2017-05-17 23:36:01.882 [error] <0.267.0> CRASH REPORT Process <0.267.0> with 1 neighbours exited with reason: {timeout,{gen_fsm,sync_send_event,[<0.267.0>,{<0.258.0>,ping},10000]}} in gen_fsm:terminate/7 line 559
2017-05-17 23:36:01.883 [error] <0.257.0>@lorawan_connector_factory:handle_exit2:131 Connector DemoConnector terminated: {timeout,{gen_fsm,sync_send_event,[<0.267.0>,{<0.258.0>,ping},10000]}}
2017-05-17 23:36:01.883 [debug] <0.256.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"esmqdev1.zedisolutions.com",1883,"/",[]}, {connector,<<"DemoConnector">>,true,<<"mqtt://esmqdev1.zedisolutions.com:1883">>,<<"mqtt/zedimote/...">>,...}) at pid <0.2119.0>
```

This morning I tried to reconnect, and received
```
2017-05-19 09:28:20.795 [error] <0.257.0>@lorawan_connector_factory:init_connector:111 Cannot connect <<68,101,109,111,67,111,110,110,101,99,116,111,114>>: {error,already_present}
```

The initial mqtt failure (two days ago, on the 17th) generated a crash report (attached)
[crash may 17.txt](https://github.com/gotthardp/lorawan-server/files/1014772/crash.may.17.txt)

gotthardp, 2017-05-20:
Hmm. I reworked this a bit. The http://swimgate.eu/lorawan-server-0.4.1.tar.gz should behave better.

gotthardp, 2017-05-24:
And this should be fixed too now.

closed 2017-05-24 by gotthardp