gotthardp/lorawan-server #97: Crash on invalid JSON downlink data that is numeric

DigitalPigeon, 2017-05-15:
Able to reproduce the following crash on posting downlink data via mqtt

Set the handler format to JSON

Publish a message (I am using the new {deveui} identifier to route downlink messages), NOT in JSON format, with a payload of
```
Test Message
```

And lorawan-server will correctly block the data with the following error
```
10:57:38.761 ←[1;31m[error] Bad downlink json_syntax_error←[0m
```

However, if you publish a numerical value as payload (not a JSON object), for example
```
123456
```

And this caused a crash dump
```
10:57:45.817 ←[1;31m[error] gen_server <0.260.0> terminated with reason: no function clause matching lorawan_admin:parse(123) line 31←[0m
[warning] [Client <0.639.0>] Connection lost for: tcp_closed
10:57:45.834 ←[1;31m[error] CRASH REPORT Process <0.260.0> with 0 neighbours exited with reason: no function clause matching lorawan_admin:parse(123) line 31 in  gen_server:terminate/7 line 812←[0m
```

I have attached the applicable crash log
[crash.txt](https://github.com/gotthardp/lorawan-server/files/1001833/crash.txt)

gotthardp, 2017-05-15:
MQTT is a new functionality, so it needs some rough testing. Thanks for that!

gotthardp, 2017-05-15:
I believe I fixed it here: http://swimgate.eu/lorawan-server-0.4.1.tar.gz

DigitalPigeon, 2017-05-18:
Sorry Petr, I thought I commented on this yesterday.

I tested this a number of ways with the intention of breaking it, and was unable to. It looks like your fix works perfect. Thank you!

closed 2017-05-18 by DigitalPigeon

gotthardp, 2017-05-18:
Thanks for the confirmation.