gotthardp/lorawan-server #98: Sending downlink messages

Sviests, 2017-05-16:
Hi!
I do feel a bit silly, especially when I see that it works for other people, but I still can`t manage to send downlink messages to any of my nodes...

I am receiving the uplinks and they are posted to the respective MQTT topics in JSON, but when I try to post JSON data to the Consumed topic of my MQTT connector, nothing happens and nothing is written in the logs either...

An example of a message I am trying to send:
Connector consumed topic - water_meter/in/{devaddr}
the {devaddr} is AA6DDECC (I receive it in the uplink messages)
A message I am trying to send - {"data":"00", "devaddr":"AA6DDECC", "time":"2017-05-17T13:50:22Z"}

There`s nothing in the logs on any failures, the downlink counter for the node does not increase and I don`t see any scheduled downlink messages in the Node section...
I`ve played around with the JSON fields I am including in the message, but nothing seems to change...

gotthardp, 2017-05-16:
Oh, when you include "time" it becomes a Class C operation. Could you omit "time" please and see how it works?

Sviests, 2017-05-16:
No difference...

gotthardp, 2017-05-16:
Do you see the MQTT message being consumed by the server?

Sviests, 2017-05-16:
Do you mean in the logs? There's nothing in the logs after I`ve posted a downlink message...
Also, I do get quite a few errors for the MQTT connectors if I`ve change anything in their settings. After restarting the server, the errors are gone. An example of an error:
2017-05-16 12:32:15.041 [error] <0.571.0>@lorawan_connector_factory:init_connector:111 Cannot connect <<77,81,84,84,32,84,101,115,116,32,78,111,100,101>>: {error,{already_started,<0.355.0>}}
2017-05-16 12:32:15.042 [error] <0.1196.0>@lorawan_connector_factory:publish:24 Error {already_started,<0.355.0>}

DigitalPigeon, 2017-05-16:
Hi Sveists,

I'm no expert by any stretch, but I've noticed that if you are using queue names to route data, then don't include the devaddr or deveui in the payload. 

So instead, if you subscribing to:
**water_meter/in/#**

and you are consuming as:
**water_meter/in/{devaddr}**

Then your app should publish an MQTT message to **water_meter/in/AA6DDECC** with a payload of **{"data":"00"}**

As for the errors you noticed, my experience with the initial mqtt support in lorawan-server has two important notes

- You can't change the connector settings live. I find I always need to deactivate and reactive the connector.
- Once you get an error on the mqtt connector, loarwan-server requires a restart to restore connection (or you will keep getting the **already_starterd** error on the connector factory.

Don't know if any of that will help you out.

Sviests, 2017-05-17:
The behaviour with MQTT is exactly what I was having and I too found out that if you disable them first, before editing, you can avoid the errors, thanks for confirming that I was not the only one having these issues!

Your suggestions for sending downlink messages worked brilliantly, thank you so much for clearing that mystery up for me! :D

gotthardp, 2017-05-20:
Please try the nightly build http://swimgate.eu/lorawan-server-0.4.1.tar.gz; it should behave better.

gotthardp, 2017-05-24:
Let me close this issue too (fee free to comment/reopen) if you disagree.

The downlink messages seem to be working and in 0.4.2 I fixed both issues mentioned above: the already_started issue as well as the change of subscribed topics in run-time, without connector re-activation.

closed 2017-05-24 by gotthardp