gotthardp/lorawan-server #99: Tx power jumping

Kralg, 2017-05-16:
Hello,

I have noticed that if I switch between 20 and 19dBm TX power, actually power jumps much more than that. Is there a way to make set 18-19dBm?

gotthardp, 2017-05-16:
You mean uplink or downlink power? For uplink it is not possible to set any other value than those in the web-admin list.

Kralg, 2017-05-22:
Well I mean the ones that are in web-admin TX Power(dBm) I can see that I can increment the power by 1dBm. I connect Gateway output to frequency analyzer (no antenna) and I can see that actual power at 19dBm is much less and at 20dBm it jumps to somewhere closer to 20dBm.

gotthardp, 2017-05-22:
The TX power you set corresponds to the power sent by LoRa chip. The attenuation depends on the HW design of the signal path between the LoRa chip and your measuring equipment. Apparently the attenuation is not linear, but this is nothing related to the server.

Kralg, 2017-05-29:
So you think the power settings are not related to lut in global_conf.json? 
I thought that webserver sends command to packet forwarder which then looks for configurations in global_conf.json and choose corresponding ones. If there is none exactly matching then it (probably) takes the closest ones. I was looking on ttn webpage  and I found some similar discussions just without clear explanation on how to create intermediate power settings lut, because luts are just for some of the power settings but not for all settings with 1dBm increment.

gotthardp, 2017-05-29:
To be honest, I was not aware of the `tx_lut_xx` configuration. :) For downlinks the server sends directly the TX power in dBm. I don't know what the lut is good for.

gotthardp, 2017-07-12:
Did you manage to solve this somehow? May I close this issue?

closed 2017-07-21 by gotthardp