gotthardp/lorawan-server #100: Server Installation

suhaibfs, 2017-05-17:
Hello there,
Whenever I run bin/lorawan-server I get this response:
````````````````````````
~/lorawan-server# bin/lorawan-server
{"init terminating in do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{{badmatch,{error,prerequisite_failed}},[{lorawan_app,start,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,19}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,273}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_it,1,[]},{init,start_em,1,[]}]}}

Crash dump is being written to: erl_crash.dump...done
init terminating in do_boot ()
````````````````````````
I used [Using the Binary Release on Linux] method in installing the server.
What could possibly be the issue here?
Thanks

gotthardp, 2017-05-17:
Hello. I guess you have some older Erlang. The latest server required 19+

suhaibfs, 2017-05-17:
Thank you for your quick reply,
Issue solved.

closed 2017-05-17 by suhaibfs