gotthardp/lorawan-server #101: Starting lorawan-server problem

Gligor721, 2017-05-18:
When I start lorawan-server I get:

[error] <0.116.0> CRASH REPORT Process <0.116.0> with 0 neighbours exited with reason: no match of right hand value {error,prerequisite_failed} in lorawan_app:start/2 line 19 in application_master:init/4 line 134
 What is the problem?

gotthardp, 2017-05-18:
Erlang 19+ is required. You have something older.

Gligor721, 2017-05-18:
I solve that problem but now I received message from my device but how to send message to device?I try with demo webclient and connect to my group TEST and send row message but in the counters of the device down counter is 0.In the node i saw that I generate message but is not send!What I need to configure so i can send message?

gotthardp, 2017-05-18:
is there something in the server logs?

Gligor721, 2017-05-19:
I solved the problem with sending the message to remote node but now I have another question: Can server send all message to remote node when server received message from node not to send only one message but to send all message from down-link buffer?

gotthardp, 2017-05-19:
Could you share with me (and other readers) where was the problem? Is there a way to avoid this by an update of documentation, for example?

The node opens only one window for one message. However, there is a flag for a device to indicate there are other messages to send, so the device can just send another message and open another windows for the second frame.

Gligor721, 2017-05-19:
Hi,

the problem is that I set the application on the device to web socket and with the demo I connect to the remote device and received the message and send message from server to remote device all is ok but i saw that when remote node send message server send one message from downlink buffer and I add multiple message to the buffer.Can server when received message  from node send all message to remote node?

Another question: How to open the server database so i can analyze received message?

gotthardp, 2017-05-19:
Per LoRaWAN standard there is one downlink (server to node) slot after each uplink (node to sever), so the answer is no, there is no way to send multiple messages to one node at one time. You should consider combining the messages into one. ---or (and that may be an interesting feature) the server can auto-delete the old messages and send always the latest received from the application.

You can access the last 50 received frames via the REST API, see "rxframes" in https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md.

gotthardp, 2017-05-20:
I clarified `prerequisite_failed` in the https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md

closed 2017-05-20 by gotthardp