gotthardp/lorawan-server #102: Missing uplink msgs - Problem in frequency channel setting in Gateway

kammyy, 2017-05-25:
Hi All,

I am missing many uplinks sent from the device. There is something wrong with the frequency channel setting in the gateway configuration file.
My device is working on following frequencies:
a.    Channel 0 = 865.0625 MHz (SF 7 to 12, BW125kHz)
b.   Channel 1 = 865.4025 MHz (SF 7 to 12, BW125kHz)
c.   Channel 2  = 865.9850 MHz (SF 7 to 12, BW125kHz)
d.   RX2 = 866.5500 MHz (SF8, BW 125kHz)

I am using Multi-tech conduit gateway. 

Please help in correcting my configuration file so that the gateway is listening on the above mentioned frequencies.

Below is the gateway conf file that i am using - 

`{
    "SX1301_conf": {
        "lorawan_public": true,
        "clksrc": 1,
        "lbt_cfg": {
            "enable": false,
            "rssi_target": -80,
            
            "sx127x_rssi_offset": -4
        },
        "antenna_gain": 0,
        "radio_0": {
            "enable": true,
            "type": "SX1257",
            "freq": 865500000,
            "rssi_offset": -166.0,
            "tx_enable": true,
            "tx_notch_freq": 129000,
            "tx_freq_min": 863000000,
            "tx_freq_max": 870000000
        },
        "radio_1": {
            "enable": true,
            "type": "SX1257",
            "freq": 866500000,
            "rssi_offset": -166.0,
            "tx_enable": false
        },
        "chan_multiSF_0": {
            "enable": true,
            "radio": 1,
            "if": -1437500
        },
        "chan_multiSF_1": {
            "enable": true,
            "radio": 0,
            "if": -97500
        },
        "chan_multiSF_2": {
            "enable": true,
            "radio": 0,
            "if": 485000
        },
        "chan_Lora_std": {
            "enable": true,
            "radio": 1,
            "if": 50000,
            "bandwidth": 125000,
            "spread_factor": 8
        },
        "chan_FSK": {
            "enable": true,
            "radio": 1,
            "if": 300000,
            "bandwidth": 125000,
            "datarate": 50000
        }
    },

    "gateway_conf": {
        "gateway_ID": "000800FFFF4A166E",
        "server_address": "localhost",
        "serv_port_up": 1680,
        "serv_port_down": 1680,
        "keepalive_interval": 10,
        "stat_interval": 30,
        "push_timeout_ms": 100,
        "forward_crc_valid": true,
        "forward_crc_error": false,
        "forward_crc_disabled": false
    }
}`

Thanks

gotthardp, 2017-05-25:
Hello, I am not the gateway expert, so I cannot help you there, however please note that default RX2 for the EU band is 869.525 / SF12BW125, so you definitely need to modify that in the **server config** in order to receive RX2 transmissions.
Add something like this into your config
```erlang
{regions, [
        {<<"EU863-870">>, [{rx2_rf, {866.55, <<"SF8BW125">>}}]},
]},
```

kammyy, 2017-05-26:
I am using 865 to 867 Mhz band. Also, since i am only missing uplinks, so there cant be an issue with RX2 frequency.
I am not entirely sure of the reasons as to why i am missing almost 60% of the uplinks. The best possible deduction is that it is due to the frequency channel setting in the gateway.

What else could be wrong in case i am missing lot of uplink messages?

Please suggest appropriate solutions.

gotthardp, 2017-05-26:
I can only state the obvious: if 2/3 of the uplinks are lost, then likely two of the 3 frequencies are badly configured either in the device or in the gateway. It may be because either the device or the gateway cannot handle the .xx25 precision of the frequency setting. However this is my guess only.

kammyy, 2017-05-29:
Problem solved. Now able to receive all the uplinks.

gotthardp, 2017-05-29:
Cool. What was the root cause, please?

closed 2017-05-29 by gotthardp

susiloharjo, 2018-05-24:
Can you tell me fix that @kammyy ?