gotthardp/lorawan-server #103: Error starting server.

gjoosen, 2017-05-26:
When I'm running the server I get the following error:

2017-05-26 12:09:07 Can't set short node name!
Please check your configuration

Both configurations are default, but I can't find a short node name, only a in the lorawan-forwarder.xml, but changing that short doesn't work. I'm running the server through the MacOS terminal, running on the latest version of MacOS.

gotthardp, 2017-05-26:
Hello. It means the hostname of your machine is not "properly" configured. It has something to do with results of `hostname --short` and `hostname --long`. Would you be able to set the hostname of your machine to have both a long and a short hostname?

gjoosen, 2017-05-27:
Hello. Changing the hostname to a simple name without weird characters solved the problem. On MacOS you can do this by applying the following command in the terminal: `sudo scutil --set HostName newHostname`.

closed 2017-05-27 by gjoosen

gotthardp, 2017-05-27:
Cool! Thanks for letting me know.