gotthardp/lorawan-server #104: Downlink messages not sent

webfrank, 2017-05-26:
Hi, I have setup the server and I'm receiving data via MQTT. When I try to send a downlink packet like this:

`{
	"devaddr": "00000001",
	"fields": {"led": 1},
	"port": 1
}`

I could see it on Node page listed as downlinks, and when the node transmits it got deleted but I cannot receive the data neither I see the tx packet on the gateway, it seem the server deletes the packet but never transmits it.

gotthardp, 2017-05-26:
The downlink gets removed from the list, but you don't see any TX in the gateway log, correct? That is weird. Do you see any errors in the server log?

webfrank, 2017-05-27:
Hi, yes it is and the weird thing is no udp packet is sent out from the server. I'm using the docker version with latest tag. I sniffed the server traffic with tcpdump.

gotthardp, 2017-05-27:
Would you please be able to help me debugging this. It looks like a bug, but I cannot reproduce it here.

First of all, could you please send me (post here the tail of) the server `error.log`?

webfrank, 2017-05-27:
Hi, I restarted the server, the gateway and the node and it seems it was working ok, I didn't received the packet but I saw it on the network and on the radio, probably a node problem (anarduino + lmic) but after some time of testing the server does not send anymore downlink packet to the gateway. error.log and debug.log does not report anything.
I tried to stop only the server, wait some time and restart it and for some time again it talks to the gateway.

gotthardp, 2017-05-27:
This is even harder to debug. There must be something in the log or some other sign of what could happened.

webfrank, 2017-05-28:
I'm trying to install directly on linux to exclude docker issues. Let me know which kind of test I could do. I suspect something on the packet forwarder code stop at some time because the gateway stops receiving other types of packets like PUSH_ACK and PULL_ACK. Receiving of data from gateway seems not affected.

gotthardp, 2017-05-28:
Data are sent and received by the same process, so **if** the server receives frames from the gateway, but the gateway stops receiving all ACKs, then I suspect something is wrong at the network/UDP layer.

The best is to look in the logs (debug.log). There must be:
 * Exactly one `gateway xxx connected` message
 * No `xxx terminated` message

Anything else is suspicious and could be a sign of a failure. Could you please double check?

webfrank, 2017-05-28:
Hi, just pulled latest docker image, now I got this error with the same configuration but data from node is arriving:

2017-05-28 07:52:24.621 [debug] <0.313.0>@lorawan_mac_commands:build_fopts:32 <<0,0,0,1>> <- [dev_status_req]
2017-05-28 07:52:24.622 [error] <0.313.0> node 00000001 {disabled,<<77,81,84,84>>}

webfrank, 2017-05-28:
Hi,I pulled latest docker image and now I have this error (probably why I'm not receiving the data on the node, but with SDR I can see gateway is transmitting):

2017-05-28 08:00:16.740 [warning] <0.262.0> gateway 5CCF7FF42F2CD6BE {downlinks_lost,1}

gotthardp, 2017-05-28:
This is just a warning that gateway did refuse one downlink. In the gateway log there may be a reason for that.

webfrank, 2017-05-28:
Ok, after one hour is still working, probably previous "latest" image got some problem just resolved.

Two questions:

- Is it possible to have on the left menu of the server admin a link to events list?
- How to transmit downlink on RX2?

gotthardp, 2017-05-28:
You mean between the tagged 0.4.2 and 'master'? I just did some more error logging, without fixing anything, but I am glad it works.

OK, I will add the events to the menu.

The RX window is dynamically selected based on how fast the response from the application comes. RX2 is used when it's necessary to send "late" responses. Do you need some other logic?

webfrank, 2017-05-28:
If possible to choose RX1,RX2 would be useful especially with nodes using arduino LMIC which doesn't works very well with RX1. Probably a field on MQTT json with a default of "auto" would be ok.

gotthardp, 2017-05-28:
And could you confirm this is due to a bug or some error in the device? I could happily introduce some workarounds, but I want to be sure this is not a bug in the server because if everyone switches to RX2 the (possible) bug would never get fixed.

webfrank, 2017-05-28:
No, it's not a bug, you logic is ok. The problem is with the LMIC library on arduino which does not handle well receiving on RX1. I would like to have the possibility to choose "auto" (the actual mode) or force RX1 or RX2. I think it would be a nice feature to have.

gotthardp, 2017-05-28:
OK, will do. Likely by tomorrow.

webfrank, 2017-05-28:
Tnx a lot and really nice work.

gotthardp, 2017-05-29:
Hello. In the latest 'master' image the menu contains Events (under Infrastructure) and you can choose to use RX2 only (under Device/Node *TX Window*). Please check if it works as desired.

webfrank, 2017-05-31:
Hi, just pulled latest image and is working as expected, great! I'm not receiving data but that is another story, I suspect some problem on the arduino code at this point.

Thanks

gotthardp, 2017-06-01:
I am glad it works.

closed 2017-06-01 by gotthardp