gotthardp/lorawan-server #105: Error using systemctl

gjoosen, 2017-05-28:
Hello,
I am able run the binary just fine, but when I use Systemctl I get the following error:
`lorawanctl[4049]: erlexec: HOME must be set`
Writing `echo $HOME` does print the Home directory. 
Is there something wrong with my setup?

gotthardp, 2017-05-28:
Hello. Your system is OK, this is a bug. I will fix it tomorrow.

kazi93, 2017-05-31:
@gotthardp I am receiving the same error while stopping the server. I am using the latest versing v0.4.2. Is it fixed or I am missing something. For example the now the #echo $HOME is printing /root as home directory. Does it need to be the home>> lorawan?  as per the documentation ( useradd --home-dir /var/lib/lorawan-server --create-home lorawan.

May 31 20:42:03 vps416657.ovh.net systemd[1]: Stopping LoRaWAN Server...
May 31 20:42:03 vps416657.ovh.net lorawanctl[19357]: erlexec: HOME must be set
May 31 20:42:03 vps416657.ovh.net systemd[1]: lorawan-server.service: control process exited, code=exited status=1
May 31 20:42:03 vps416657.ovh.net lorawan-daemon[17602]: Session terminated, terminating shell... ...terminated.
May 31 20:42:03 vps416657.ovh.net systemd[1]: lorawan-server.service: main process exited, code=exited, status=143/n/a
May 31 20:42:03 vps416657.ovh.net systemd[1]: Stopped LoRaWAN Server.
May 31 20:42:03 vps416657.ovh.net systemd[1]: Unit lorawan-server.service entered failed state.
root@vps416657:/var/lib/lorawan-server/log# echo $HOME
/root
root@vps416657:/var/lib/lorawan-server/log#

gotthardp, 2017-05-31:
It's not fixed yet. It's not a bug specific to this server-- The systemd ignores whatever you set; I am trying to find some standard approach how to solve this.

Sviests, 2017-06-02:
Are you sure this is a bug? I got it working perfectly following the instructions of the binary release install, just remember to make a lorawan user like so `useradd --home-dir /var/lib/lorawan-server --create-home lorawan`

gotthardp, 2017-06-02:
Its a minor issue only. But the home is not set error shoulnt be there

gotthardp, 2017-06-02:
Now its fixed. Will be included in the release 0.4.3 which is about to be released tomorrow.

closed 2017-06-02 by gotthardp