gotthardp/lorawan-server #106: Dowlink for Class A

kammyy, 2017-05-29:
While sending the downlink message for Class A devices, the json recommended is -

{"devaddr":"11223344", "data":"0026BF08BD03CD35000000000000FFFF", "confirmed":true}
    {"data":"00", "port":2, "time":"2017-03-04T21:05:30.2000"}

For A type device, the downlink is sent immediately after previous uplink.

1. Why do i need to specify "the time" in the json above?
2. How am i supposed to know the time when the next uplink from the A type device will come?

Thanks

gotthardp, 2017-05-29:
Hello. I hope I didn't say anywhere this was a Class A example. You're right, you cannot know the uplink time-- the "time" parameter is for Class C only (as indicated [here](https://github.com/gotthardp/lorawan-server/blob/master/doc/JSON.md)).

kammyy, 2017-05-29:
It means that the jason for class A is:
{"devaddr":"11223344", "data":"0026BF08BD03CD35000000000000FFFF", "confirmed":true}

What about port in class A?

Also,for class C: What is the point of having time . Isn't the message sent to device immediately by default?

gotthardp, 2017-05-29:
Correct.

Have a look in the documentation of the `port` parameter: *If not specified for Class A, the port number of last uplink will be used. Mandatory for Class C.*

The `time` is to distinguish between Class A and Class C downlinks. Even class C device can do Class A downlinks.

kammyy, 2017-05-29:
Understood.

Thanks Petr.

gotthardp, 2017-05-29:
Glad I could help :) I also clarified the documentation.

closed 2017-05-29 by gotthardp