gotthardp/lorawan-server #107: Can this Server be used for Commercial Grade Solutions?

kammyy, 2017-05-29:
Dear Petr,

I was just wondering, Can this LoRaWAN server be used to deploy commercial private lorawan networks?

 How many gateways and devices can it practically support? Is there any maximum limit on the number of gateways and devices?

We are planning to use this server for a Smart Community project. We have about 3-4 gateways and about 1000 end nodes.

Thanks

gotthardp, 2017-05-29:
Well, I intend to use this server for a deployment with similar number of gateways and about 6000 nodes, so I do everything to make it work.

Currently the server is missing some network monitoring features (issue #85), is lacking provisions for assuring regulatory compliance (issue #84), there may be some performance and reliability issues and the web-admin may have troubles handling this amount of nodes, but my plan for the coming 6 months is to fix all that.

kammyy, 2017-05-29:
Well, that sounds great. Hope we both start the commercial solutions on this server soon.

Thanks

closed 2017-05-31 by kammyy