gotthardp/lorawan-server #108: Server not sending ACK for confirmed uplinks from device.

kammyy, 2017-05-29:


kammyy, 2017-05-29:
Sorry A typo:

My end nodes are not receiving any ACK from "SERVER" against confirmed up-links.

Thanks

gotthardp, 2017-05-29:
Does this mean that you don't get any downlinks. or just acks?

kammyy, 2017-05-29:
Did not tried the downlink yet.

gotthardp, 2017-05-29:
Some devices like the arduino LMIC have problems receiving downlinks in the RX1 window, which I use. I will make a workaround for them, which may improve things also for you.

kammyy, 2017-05-29:
But we have used loriot for some time and our devices work very well with Loriot Network Server.

Are you sure that there cant be an issue on the server side?

kammyy, 2017-05-29:
Also, if the device is not listening on the RX1, then should it not listen on RX2 ?

kammyy, 2017-05-29:
Is there a way to check whether the server sending the ACK to the device or not?

gotthardp, 2017-05-29:
There can always be a bug in the server, but in my environment all ACKs are working 100%, so I need help with debugging. I checked and re-checked everything and I cannot see what it doesn't work for some, but works for others.

The best way to check server sent something is to look in the gateway log. There should be a downlink related record.

kammyy, 2017-05-29:
I dont find any downlink log in the gateway,

In the communication.md file, it is mentioned:

1.   "The server uses the standard downlink frequencies and data rates." : What are these standard frequencies and data rates?

I am using version 0.4.1, i the current "lorawan-server/releases/<VERSION>/sys.config" file, region wise RX2 frequency is not given. 
When i try to add the
 `{regions, [
    {<<"EU863-870">>, [
        % default RX2 frequency (MHz) and data rate (DRx)
        {rx2_rf, {869.525, <<"SF12BW125">>}}
    ]},` 
in the sys.config file the i get the following error while starting the server:
"could not start kernel pid (application_controller) (invalid config data: application: regions; invalid parameter: <<"EU863-870">>)"

Please help here.

Thanks

gotthardp, 2017-05-29:
The default values are given in https://github.com/gotthardp/lorawan-server/blob/master/src/lorawan_server.app.src

You probably added `regions` at the top level. It must be among other parameters in the `lorawan_server` section.

kammyy, 2017-05-30:
Successfully changed the sys.config for my desired rx2 frequency.

Still no luck with ACK from server or any downlink message.
Will keep you updated. 
Now , when i send the downlink (for a C type device, with time parameter as immediately), it is shown as queued downlink in the http://127.0.0.1:8080/admin#/nodes/edit/0040D0E3 section.

Please suggest any-other possible changes for ack and downlink.

gotthardp, 2017-05-30:
If it's queued then the "time" parameter was not recognized. Are you sure there is no typo?
Also, could you upgrade to latest 'master' and set 'TX Window" for your Device/Node to "RX2"?

kammyy, 2017-05-30:
I am in the time zone GMT +5:30. Please suggest the format to follow for time.

I have downloaded the latest server version (0.4.2) but i dont an option to set 'TX Window' anywhere.

kammyy, 2017-05-30:
Sorry a typo:

"but i dont SEE an option to set 'TX Window' anywhere"

gotthardp, 2017-05-30:
The 'TX Window' option is in the 0.4.3, which is not released yet.

kammyy, 2017-05-30:
I have taken the latest master, but still no option of TX window.

How to get your latest code?

gotthardp, 2017-05-30:
It is there. Please try refreshing your browser.

kammyy, 2017-05-30:
Got it.

Will keep you updated regarding the downlink.

kammyy, 2017-05-30:
On the gateway logs, i am getting the following error:

"ERROR: INVALID RF_CHAIN TO SEND PACKETS"

What it could mean?  How to remove it?

thanks

kammyy, 2017-05-30:
I was using channel "2". When i changed it to "0" then the downlink is working but the uplink is not working now. 
What exactly is "TX Chain" for a gateway and what does its value signifies?

thanks

gotthardp, 2017-05-30:
Oh, that was wrong. The "channel" relates to gateway's `global_conf.json`: there is `radio_0`, `radio_1`, perhaps more and some have `tx_enable: true` and some not. The channel number shall be set to radio_x number whose tx is enabled.

kammyy, 2017-05-30:
Uplink and downlink working fine!!!!!
Thanks for answering all my silly questions :))

Now i will add multiple end nodes as well as gateways and will share the test results soon.

Thanks again

gotthardp, 2017-05-30:
Glad it works. What were the errors you fixed? The channel settings, what else? Could you summarize your findings please?

kammyy, 2017-05-30:
For downlink communication, i made following changes:

1. "TX Chain" setting in gateway config. Set the "TX chain" value to "0" because "tx enable" is true for radio_0 in my gateway config file.

2. Selected RX2 for TX Window.

Just these solved the uplink and downlink problems.

gjoosen, 2017-05-31:
Where is it possible to set the TX Window?

gotthardp, 2017-06-01:
I am glad it works.

closed 2017-06-01 by gotthardp

ashok410, 2018-07-05:
Hi,
my end device not getting downlink messages from MQTT, but it gets automatic downlinks when a uplink packet is received(which is not scheduled by me from MQTT)
please give me sugesstion
Thanks in Advance
Regards,
Ashok

gotthardp, 2018-07-05:
Dont reuse old issues please.