gotthardp/lorawan-server #109: Getting "disabled" error

felixgalindo, 2017-05-30:
Keep getting error shown below after upgrading to 0.4.2.

"applications">>],[],lorawan_admin_applications,...},...]}]}}) at pid <0.232.0>
06:39:20.482 [info] server started
06:39:20.509 [debug] Supervisor lorawan_sup started lorawan_gw_router:start_link() at pid <0.247.0>
06:39:20.520 [debug] Supervisor lorawan_sup started lorawan_gw_forwarder:start_link([{port,1680}]) at pid <0.248.0>
06:39:20.524 [debug] Supervisor lorawan_sup started lorawan_connector_sup:start_link() at pid <0.249.0>
06:39:20.529 [debug] Supervisor lorawan_sup started lorawan_connector_factory:start_link() at pid <0.250.0>
06:39:20.529 [info] Application lorawan_server started on node lorawan@raspberrypi
06:39:28.386 [info] gateway B827EBFFFEFA0134 {connected,{{127,0,0,1},45593}}
06:40:01.512 [debug] <<87,255,187,75>> <- [dev_status_req]
06:40:01.541 [error] node 57FFBB4B {disabled,<<77,81,84,84>>}
06:41:03.780 [debug] <<87,255,187,75>> <- [dev_status_req]
06:41:03.783 [error] node 57FFBB4B {disabled,<<77,81,84,84>>}

gotthardp, 2017-05-30:
This means "connector disabled". It's not a bug, it's a feature. You likely set application to "backend", set some connector, but disable it, so the frames cannot be forwarded. Could this be the case?

gotthardp, 2017-06-01:
I clarified this in documentation.

closed 2017-06-01 by gotthardp