gotthardp/lorawan-server #110: Unable to send data when integrating over mqtt

Ankitjd18, 2017-05-30:
Hi, I followed the integration doc to integrate with generic mqtt, but I am getting below error

2017-05-30 14:01:02.565 [error] <0.1442.0> gen_server <0.1442.0> terminated with reason: bad argument in gen_fsm:send_event/2 line 215
2017-05-30 14:01:02.565 [debug] <0.1508.0>@lorawan_connector_mqtt:init:23 Connecting test to mqtt://127.0.0.1:1883
2017-05-30 14:01:02.566 [error] <0.1442.0> CRASH REPORT Process <0.1442.0> with 0 neighbours exited with reason: bad argument in gen_fsm:send_event/2 line 215
2017-05-30 14:01:02.567 [error] <0.264.0> Supervisor lorawan_connector_sup had child <<"test">> started with lorawan_connector_mqtt:start_link({mqtt,[],"127.0.0.1",1883,"/",[]}, {connector,<<"test">>,true,<<"mqtt://127.0.0.1:1883">>,<<"out/0040D0E3">>,<<>>,<<"in/0040D0E3">>,...}) at <0.1442.0> exit with reason bad argument in gen_fsm:send_event/2 line 215 in context child_terminated
2017-05-30 14:01:02.567 [debug] <0.264.0> Supervisor lorawan_connector_sup started lorawan_connector_mqtt:start_link({mqtt,[],"127.0.0.1",1883,"/",[]}, {connector,<<"test">>,true,<<"mqtt://127.0.0.1:1883">>,<<"out/0040D0E3">>,<<>>,<<"in/0040D0E3">>,...}) at pid <0.1508.0>
2017-05-30 14:01:02.567 [error] <0.1444.0> gen_server lorawan_connector_factory terminated with reason: no match of right hand value {error,running} in lorawan_connector_factory:handle_exit/2 line 141

Please help, I am unable to figure out the issue.

gotthardp, 2017-05-30:
This is clearly a bug. Could you please send me the entire log, or at least few more lines of what happened before?

Ankitjd18, 2017-05-30:
It is working now, I must have missed something. Thank you for your reply.
Now, I am trying to publish message through mqtt but getting error as - Unknown downlink target: #{}
My connector setting is as below
[image]
I am publishing to topic - "in/0040D0E3".
Can you please help me here, what I am doing wrong.

gotthardp, 2017-05-30:
"Unknown downlink target" means that you didn't specify anywhere to what device the downlink shall be sent. You have two options:
 - either define a "devaddr" or "deveui" field in the JSON structure
 - or define the "Consumed Topic" as a template "in/{devaddr}", which causes the server to parse the topic and extract the DevAddr from there

Ankitjd18, 2017-05-31:
I have defined the "Consumed topic" same as you mentioned in the comment, you can check the same in the attached image in above comment. Also, added a devaddr field in JSON. Json structure as follows -
{"devaddr":"0040D0E3", "data":"00", "confirmed":true, "port":2, "time":"immediately"}

gotthardp, 2017-05-31:
The screenshot shows a number after `in`. What i mean is there should be the exact text I recommended. Not replaced by a number.

Ankitjd18, 2017-05-31:
Thanks changed the same, now I am getting below error -
2017-05-31 12:02:38.681 [error] <0.2597.0>@lorawan_connector_mqtt:match_vars:243 Topic <<105,110,47,123,100,101,118,97,100,100,114,125>> does not match pattern {<<105,110,47,48,48,52,48,68,48,69,51>>,[]}
2017-05-31 12:02:38.681 [error] <0.2597.0>@lorawan_application_backend:send_downlink:147 Unknown downlink target: #{}

gotthardp, 2017-05-31:
The cryptic error says the topic <<"in/{devaddr}">> does not match pattern <<"in/0040D0E3">>. This sound like you kept the "Consumed Topic" the same and you send MQTT topic <<"in/{devaddr}">>. Could this be the case?

Ankitjd18, 2017-05-31:
I have changed the consumed topic to **"in/{devaddr}"** then also getting the error. Tried restarting the server also. My exact request is below -
`mosquitto_pub -h 127.0.0.1 -p 1883 -t "in/{devaddr}" -m "{"devaddr":"0040D0E3", "data":"00", "confirmed":true, "port":2, "time":"immediately"}"  -u "ankit" -P "ankit"`
And my connector settings is bellow -
[image]

gotthardp, 2017-05-31:
Connector settings are OK, but the `-t` parameter shall be `-t "in/0040D0E3"`

Ankitjd18, 2017-05-31:
Thank you for your quick response. Earlier, I tried using as suggested also, but it did not work at that time. Maybe server restart did the change. Now, I am getting `json_syntax_error` but my JSON seems fine and I haven't added any JSON in connector or handler.

gotthardp, 2017-05-31:
JSON error means JSON error. I suspect its because the quotation marks (you cannot use `"` if you need to send `"`). I suggest
`mosquitto_pub -h 127.0.0.1 -p 1883 -t 'in/{devaddr}' -m '{"devaddr":"0040D0E3", "data":"00", "confirmed":true, "port":2, "time":"immediately"}' -u 'ankit' -P 'ankit'`

Ankitjd18, 2017-05-31:
It's working now. Thank you for all your help.

closed 2017-05-31 by Ankitjd18