gotthardp/lorawan-server #112: time to receive !!

YousseMes, 2017-05-30:
Hi !
thank you for your Job !
I have just one probleme why I receive date every minute and not in really time
please thear's some configuration to fix that ?
[image]

gotthardp, 2017-05-30:
Hi. I am not sure I understand the issue. The device sends the data. It's upto the device to decide when to send data. What do you expect to see there?

YousseMes, 2017-05-30:
I want to receive in server at real time

gotthardp, 2017-05-30:
LoRaWAN is not intended for real-time operations. The devices have to obey duty cycle regulations, so they have to wait few seconds before sending again.

Ankitjd18, 2017-06-01:
Hi,
Below is the JSON payload we receive on uplink
`{"data":"0E5E3345395B6D84","datetime":"2017-06-01T09:33:18Z","devaddr":"0040D0E3","fcnt":270,"group":"testing","last_lost":false,"port":2,"rxq":{"codr":"4/5","datr":"SF12BW125","freq":865.403,"lsnr":9.2,"rssi":-27,"time":"2017-06-01T09:34:33.308854Z","tmst":1530369108},"shall_reply":true}`

Here we can see fields as "datetime", "time", which has a constant difference of 1 min 16 sec. What are these two "time" fields? Also, what is `tmst` field in the payload?

gotthardp, 2017-06-01:
Hi. The `rxq` parameters correspond to what the gateway set. See https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT. The `datetime` is a server time, whereas the `time` is GPS time.

gotthardp, 2017-06-01:
Will update the documentation accordingly.

closed 2017-06-01 by gotthardp