gotthardp/lorawan-server #114: Received frames log

Sviests, 2017-06-02:
Hi! Is there a limit to how big the received frames log can get? I am having issues with frames disappearing from the log and the log not getting larger than 4 pages in the WebUI. For example, I have a node which reports hourly and after some amount of received frames, the older ones started to disappear...

Also, the export function in the received frames view. Is there a possibility that the exported file would be sorted by date or so that the user could specify how it should be sorted?

gotthardp, 2017-06-02:
Hello. There is a `retained_rxframes` parameter, which is now more public. It defines how many frames are retained. You can increase this number if you want.

The export function doesn't filter, but you can use the REST API, which supports filtering and just store the JSON to a file. See https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md. Would this be an option for you?

Sviests, 2017-06-02:
This works for me, thanks!

closed 2017-06-02 by gotthardp