gotthardp/lorawan-server #115: The server started but shut down by  packet_forwarder interface

suyiii, 2017-06-03:
Hello!Thank you for your project.
I compiled the project and released it.In the debian over Raspberry Pi, when I started the packet-forwarder service and executed "./bin/lorawan-server", it failed, and the debug message is:
03:18:33.736 [info] server started
03:18:33.763 [debug] Supervisor lorawan_sup started lorawan_gw_router:start_link() at pid <0.239.0>
03:18:33.773 [error] Failed to start the packet_forwarder interface: ~w
03:18:33.774 [info] gateway router terminated: shutdown
03:18:33.774 [error] CRASH REPORT Process <0.240.0> with 0 neighbours exited with reason: eaddrinuse in gen_server:init_it/6 line 344
03:18:33.775 [error] Supervisor lorawan_sup had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at undefined exit with reason eaddrinuse in context start_error
03:18:33.776 [error] CRASH REPORT Process <0.204.0> with 0 neighbours exited with reason: {{shutdown,{failed_to_start_child,packet_forwarder,eaddrinuse}},{lorawan_app,start,[normal,[]]}} in application_master:init/4 line 134
03:18:33.777 [info] Application lorawan_server exited with reason: {{shutdown,{failed_to_start_child,packet_forwarder,eaddrinuse}},{lorawan_app,start,[normal,[]]}}

It is also very strange that I can run the project perfectly in the unbuntu over VMware but fail in the debian over Raspberry Pi.

gotthardp, 2017-06-03:
Hello. The eaddrinuse error means the UDP port 1680 is used by another application. You should shutdown that app or change the port used.

suyiii, 2017-06-03:
Thank you! Our problem resolved.
I will close the issue.

closed 2017-06-03 by suyiii