gotthardp/lorawan-server #116: Unable to start the server

Ankitjd18, 2017-06-03:
I am trying to start the server on AWS machine but getting below error - 
`{"init terminating in do_boot",{undef,[{lorawan_app,start,[],[]},{init,start_it,1,[]},{init,start_em,1,[]}]}}

Crash dump was written to: erl_crash.dump
init terminating in do_boot ()`

After browsing some of the issues here, I figured out this is due to port already in use. But nothing is running on port **8080**
Please help me here.

gotthardp, 2017-06-03:
`undef lorawan_app:start` is an installation problem. Erlang cannot find server binaries. How did you start it?

Ankitjd18, 2017-06-03:
`bin/lorawan-server` - using this, as explained in the doc.

gotthardp, 2017-06-03:
Can you see a `lib/lorawan_server-*/ebin/lorawan_app.beam` file? Is it accessible for the user starting the server?

Ankitjd18, 2017-06-03:
yes, the file is there and it is accessible to the user.
Also, after updating and proper installing, I was able to start the server. Then, I stopped the server and now I am unable to start it again. Getting below error - 
`{"init terminating in do_boot",{{badmatch,{error,{mnesia,{killed,{mnesia_sup,start,[normal,[]]}}}}},[{lorawan_app,start,0,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
init terminating in do_boot ()`

gotthardp, 2017-06-03:
Well, the only explanation I can think of is that you first run the server under root, which created database file owned by root and then you try to start it under other user, but the database files cannot be accessed as they are owned by root.

Ankitjd18, 2017-06-03:
The server has only been deployed by the same user and not root. Initially, the issue was because of not proper installation of erlang. But, now I am unable to figure out the issue.

gotthardp, 2017-06-03:
Try removing the `Mnesia.lorawan@*` directory and all its content. It should be either in `/var/lib/lorawan-server` or in the directory where you start the server.

Ankitjd18, 2017-06-03:
Thank you, it's working now. But can you explain the cause? I didn't get the issue.

gotthardp, 2017-06-04:
This was a general Erlang issue that (rarely) happens to other projects too.
I suspect that when you first started the server using an incorrectly installed erlang the mnesia database files got created incorrectly and caused the subsequent crashes. By deleting the directory you removed the faulty files and let mnesia to re-create a fresh database.

Ankitjd18, 2017-06-04:
Thank you for the explanation. It's working fine now.

closed 2017-06-04 by Ankitjd18