gotthardp/lorawan-server #117: Server send wrong rfch parameter

suyiii, 2017-06-03:
Hello!
When the device joins the network by CN470, the gateway throws an exception:
lgw_receive:1144: [7 17]
Note: LoRa packet
RX: 00 A2 00 12 89 3E 5B 24 64 40 04 30 03 20 02 10 01 0E 4E DD 07 F3 27

JSON up: {"rxpk":[{"tmst":2664789860,"chan":7,"rfch":0,"freq":472.100000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":8.8,"rssi":-80,"size":23,"data":"AKIAEok+WyRkQAQwAyACEAEOTt0H8yc="}]}
INFO: [up] PUSH_ACK received in 1 ms
INFO: [down] PULL_RESP received :)

JSON down: {"txpk":{"powe":12,"imme":false,"tmst":2669789860,"codr":"4/5","datr":"SF12BW125","freq":502.1,"modu":"LORA","rfch":8,"ipol":true,"size":17,"data":"ID6GyfbphCiX61sCqhTGwoc="}}
INFO: [down] a packet will be sent on timestamp value 2669789860
TX: 20 3E 86 C9 F6 E9 84 28 97 EB 5B 02 AA 14 C6 C2 87
ERROR: INVALID RF_CHAIN TO SEND PACKETS
WARNING: [down] lgw_send failed

You can see that in "JSON down", rfch is 8. Is it a bug? How can I correct it?

gotthardp, 2017-06-03:
Mmmm. This is weird. It's the same server, same device, same config, same time?

gotthardp, 2017-06-03:
It's a gateway configuration error. `rfch` is usually `0` (zero). You can set this in the *TX Chain* parameter of Infrastructure/Gateway configuration

suyiii, 2017-06-03:
You're right. Thank you for your patience!

closed 2017-06-03 by suyiii