gotthardp/lorawan-server #121: RX Slot used ?

kammyy, 2017-06-07:
Which RX slot is used to send ACK for confirmed uplink message for A type device?

The "TX Window" selection, on the device register page, : Is it applicable for A type device or C Type device or both?

Does the "TX Window" affects the RX slot used for sending ACK for confirmed uplink for A type device?

Thanks

gotthardp, 2017-06-07:
The Tx Window affects all. There is one algorithm for everything.

kammyy, 2017-06-07:
I am missing a lot of ACK messages when sending confirmed uplinks from A type device. What could be wrong?  

The same device did not missed ACK from server when they were configured a class C.

gotthardp, 2017-06-07:
What tx window settings you use?

kammyy, 2017-06-07:
Currently i have set it to RX1.

What should i use?

kammyy, 2017-06-07:
If i set it to Auto, what is the logic followed in selecting between two slots?

gotthardp, 2017-06-07:
Try RX2, please. "Auto" is determined based on how fast the application module responds. It usually uses RX1.

gotthardp, 2017-06-21:
Does the issue persist, or may I close this?

closed 2017-06-28 by gotthardp