gotthardp/lorawan-server #122: fcnt_gap_too_large error

kammyy, 2017-06-07:
In the events section, i can see the server is showing "fcnt_gap_too_large" for two devices (7000D004 and 7000D008). Below is the snapshot of the events section:

[image]

But, there is no FCnt miss  for those devices as shown on the Received frames page. Take a note of the time-stamp in the events section and the time-stamp in the received frame section for respective devices. Refer to below two images:

[image]

[image]

What could be possibly wrong here ?

Thanks

gotthardp, 2017-06-07:
The number after the text is the FCnt received-- it appears that the device sent a much lower FCnt than before (....008 sent 1618). This is a device bug, or something.

kammyy, 2017-06-08:
But in the lower most snapsot(which is for 008), at time stamp 23:18:42 , the device has not sent 1618, infact the device has sent 3168 . And, i dont see any Fcnt miss there.

gotthardp, 2017-06-08:
I know. The device sent 3168 and then 1618. Dont know why.

kammyy, 2017-06-08:
But i dont see fcnt 1618 on the received frame page.
If the device sent fcnt 1618,  then should not it be visible on the received frame page?

By any chance, this "fcnt_gap_too_large" is referring to the gap between Up-link and Downlink fcnt?

Thanks

gotthardp, 2017-06-08:
fcnt_gap_too_large refers to uplink fcnt.

It is not visible in the list of received frames, because the frame was rejected by the server.

gotthardp, 2017-06-21:
Do you need further assistance? May I close this?