gotthardp/lorawan-server #124: dockerhub tags

SayreBlades, 2017-06-08:
I've built a public dockerhub repo that hosts your lorawan container but with a bit more sane tagging convention that mirrors the git tags.  Would it be possible for you to maintain tag parity between github and docker?

https://hub.docker.com/r/lotik/gotthard-lorawan-server/tags/

gotthardp, 2017-06-08:
Oh, I was not aware I should be doing that. Could you please be more specific about what exactly do you expect? I am a Docker newbie, so I may be not following all best practices.

SayreBlades, 2017-06-08:
Sure!

When you tag git, you could run the following command:
```
docker build . -t gotthardp/lorawan-server:VERSION
docker push gotthardp/lorawan-server:VERSION
```

where VERSION is derived from your git tag:
```
export VERSION=`git describe --tags`
```

Something like that.  We typically do these types of things in a CI/CD script; seems like you could use a github web hook.

Using these tags makes it easy to identify which version of your server we are deploying.  In this case we need your mqtt reconnect fix, but using `stable` docker tag it wasnt clear if this container had that particular fix.

SayreBlades, 2017-06-08:
also when I built the v0.4.3 tag from your source the service crashes, so im pretty sure the stable version of your code and 0.4.3 differ somewhere:

```
{"init terminating in do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{undef,[{cowboy,start_clear,[http,10,[{port,8080}],#{env=>#{dispatch=>[{'_',[],[{[<<"applications">>],[],lorawan_admin_applications,[]},{[<<"applications">>,name],[],lorawan_admin_applications,[]},{[<<"users">>],[],lorawan_admin_database,[users,user,[name,pass]]},{[<<"users">>,name],[],lorawan_admin_database,[users,user,[name,pass]]},{[<<"gateways">>],[],lorawan_admin_database,[gateways,gateway,[mac,netid,subid,tx_rfch,tx_powe,ant_gain,desc,last_rx,gpspos,gpsalt]]},{[<<"gateways">>,mac],[],lorawan_admin_database,[gateways,gateway,[mac,netid,subid,tx_rfch,tx_powe,ant_gain,desc,last_rx,gpspos,gpsalt]]},{[<<"multicast_channels">>],[],lorawan_admin_database,[multicast_groups,multicast_group,[devaddr,region,app,appid,chan,datr,codr,nwkskey,appskey,mac,fcntdown]]},{[<<"multicast_channels">>,devaddr],[],lorawan_admin_database,[multicast_groups,multicast_group,[devaddr,region,app,appid,chan,datr,codr,nwkskey,appskey,mac,fcntdown]]},{[<<"devices">>],[],lorawan_admin_database,[devices,device,[deveui,region,app,appid,appargs,appeui,appkey,link,can_join,last_join,fcnt_check,txwin,adr_flag_set,adr_set,rxwin_set]]},{[<<"devices">>,deveui],[],lorawan_admin_database,[devices,device,[deveui,region,app,appid,appargs,appeui,appkey,link,can_join,last_join,fcnt_check,txwin,adr_flag_set,adr_set,rxwin_set]]},{[<<"nodes">>],[],lorawan_admin_database,[links,link,[devaddr,region,app,appid,appargs,nwkskey,appskey,fcntup,fcntdown,fcnt_check,txwin,last_reset,reset_count,last_rx,last_mac,last_rxq,adr_flag_use,adr_flag_set,adr_use,adr_set,rxwin_use,rxwin_set,last_qs,devstat_time,devstat_fcnt,devstat]]},{[<<"nodes">>,devaddr],[],lorawan_admin_database,[links,link,[devaddr,region,app,appid,appargs,nwkskey,appskey,fcntup,fcntdown,fcnt_check,txwin,last_reset,reset_count,last_rx,last_mac,last_rxq,adr_flag_use,adr_flag_set,adr_use,adr_set,rxwin_use,rxwin_set,last_qs,devstat_time,devstat_fcnt,devstat]]},{[<<"ignored_nodes">>],[],lorawan_admin_database,[ignored_links,ignored_link,[devaddr,mask]]},{[<<"ignored_nodes">>,devaddr],[],lorawan_admin_database,[ignored_links,ignored_link,[devaddr,mask]]},{[<<"txframes">>],[],lorawan_admin_database,[txframes,txframe,[frid,datetime,devaddr,txdata]]},{[<<"txframes">>,frid],[],lorawan_admin_database,[txframes,txframe,[frid,datetime,devaddr,txdata]]},{[<<"rxframes">>],[],lorawan_admin_database,[rxframes,rxframe,[frid,mac,rxq,average_qs,app,appid,region,devaddr,powe,fcnt,port,data,datetime,devstat]]},{[<<"rxframes">>,frid],[],lorawan_admin_database,[rxframes,rxframe,[frid,mac,rxq,average_qs,app,appid,region,devaddr,powe,fcnt,port,data,datetime,devstat]]},{[<<"handlers">>],[],lorawan_admin_database,[handlers,handler,[appid,format,fields,parse,build,connid]]},{[<<"handlers">>,appid],[],lorawan_admin_database,[handlers,handler,[appid,format,fields,parse,build,connid]]},{[<<"connectors">>],[],lorawan_admin_database,[connectors,connector,[connid,enabled,uri,published,subscribe,consumed,client_id,auth,name,pass,certfile,keyfile]]},{[<<"connectors">>,connid],[],lorawan_admin_database,[connectors,connector,[connid,enabled,uri,published,subscribe,consumed,client_id,auth,name,pass,certfile,keyfile]]},{[<<"events">>],[],lorawan_admin_database,[events,event,[evid,severity,first_rx,last_rx,count,entity,eid,text]]},{[<<"events">>,evid],[],lorawan_admin_database,[events,event,[evid,severity,first_rx,last_rx,count,entity,eid,text]]},{[<<"upload">>],[],lorawan_admin_upload,[]},{[<<"rgraph">>,devaddr],[],lorawan_admin_rxgraph,[rgraph]},{[<<"qgraph">>,devaddr],[],lorawan_admin_rxgraph,[qgraph]},{[<<"devstat">>,devaddr],[],lorawan_admin_devstat,[]},{[],[],cowboy_static,{priv_file,lorawan_server,"root.html"}},{[<<"admin">>],[],cowboy_static,{priv_file,lorawan_server,"admin/index.html"}},{[<<"admin">>,'...'],[],cowboy_static,{priv_dir,lorawan_server,"admin"}},{[<<"ws">>,type,name,<<"raw">>],[],lorawan_ws_frames,[<<"raw">>]},{[<<"ws">>,type,name,<<"json">>],[],lorawan_ws_frames,[<<"json">>]}]}]}}],[]},{lorawan_app,start,2,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,59}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,273}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
init terminating in do_boot ()

Crash dump is being written to: erl_crash.dump...done
{"init terminating in do_boot",{{badmatch,{error,{lorawan_server,{bad_return,{{lorawan_app,start,[normal,[]]},{'EXIT',{undef,[{cowboy,start_clear,[http,10,[{port,8080}],#{env=>#{dispatch=>[{'_',[],[{[<<"applications">>],[],lorawan_admin_applications,[]},{[<<"applications">>,name],[],lorawan_admin_applications,[]},{[<<"users">>],[],lorawan_admin_database,[users,user,[name,pass]]},{[<<"users">>,name],[],lorawan_admin_database,[users,user,[name,pass]]},{[<<"gateways">>],[],lorawan_admin_database,[gateways,gateway,[mac,netid,subid,tx_rfch,tx_powe,ant_gain,desc,last_rx,gpspos,gpsalt]]},{[<<"gateways">>,mac],[],lorawan_admin_database,[gateways,gateway,[mac,netid,subid,tx_rfch,tx_powe,ant_gain,desc,last_rx,gpspos,gpsalt]]},{[<<"multicast_channels">>],[],lorawan_admin_database,[multicast_groups,multicast_group,[devaddr,region,ap [... cut ...] n">>],[],lorawan_ws_frames,[<<"json">>]}]}]}}],[]},{lorawan_app,start,2,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,59}]},{application_master,start_it_old,4,[{file,"application_master.erl"},{line,273}]}]}}}}}}},[{lorawan_app,start,0,[{file,"/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_app.erl"},{line,16}]},{init,start_em,1,[]},{init,do_boot,3,[]}]}}
init terminating in do_boot ()

Crash dump is being written to: erl_crash.dump...done
```

gotthardp, 2017-06-08:
Oh, thanks for reporting this.  The root-cause for all this is that:
 * I didn't fix the `cowboy` server version in rebar and used `master`
 * The cowboy developers made an API breaking change

As a result of this, all previous tags will crash when built from sources now.

To fix this situation I have
 * released `v0.4.4`
 * synced `stable` to this
 * configured Docker to produce an image for git tags too

All the images have just been built and pushed to the Docker cloud,

SayreBlades, 2017-06-09:
Thanks @gotthardp!

gotthardp, 2017-06-09:
You're very welcome.

closed 2017-06-09 by gotthardp