gotthardp/lorawan-server #125: SSL authentication with mqtt

Ankitjd18, 2017-06-09:
Hi,
We can only see certificate and key option in the connector option. How can we connect using self-signed certificates?
Thank you in advance.

gotthardp, 2017-06-09:
You can generate a self-signed *User Certificate* (`cert.pem`) and corresponding
*Private Key* (`key.pem`) by:
```
openssl req -x509 -newkey rsa:4096 -keyout privkey.pem -out cert.pem -days 365
openssl rsa -in privkey.pem -out key.pem
```

Then simply use `mqtts://` URI and upload the two files (cert.pem, key.pem) as explained in the **Backends** documentation.

gotthardp, 2017-06-09:
[An article](https://serverfault.com/questions/836870/error-ssl3-get-client-certificatepeer-did-not-return-a-certificate-on-lighttp) suggests this may be because *The client doesn't have any valid certificate (that match with root certificates at ssl. CA-file)*, which can be easily true to self-signed certificates.

Could you try creating your own CA, signing the certs by that and supplying the CA cert to the server? There are some instructions here: https://datacenteroverlords.com/2012/03/01/creating-your-own-ssl-certificate-authority/

gotthardp, 2017-06-21:
Does the issue persist? May I close this?

Ankitjd18, 2017-06-22:
Yes, We are not able to get it working. Since we are using self-signed certificates certificate authority would be needed to make a connection with mqtt. But, here we only have the option to add certificate and key for authentication.

gotthardp, 2017-06-22:
And did you try my last recommendation? How did it went?

closed 2017-07-12 by gotthardp