gotthardp/lorawan-server #126: Unable to forward message via MQTT to AWS

frozzygoh, 2017-06-12:
Hi Gotthard,

I am currently facing some issues with the connectivity between the LNS and my AWS IoT, via the integrated backend MQTT connector application.

After successful gateway and device joining process, when a message is being sent to the LNS and attempts to be forwarded to AWS MQTT, in the logs, i get the following debug logs (see below).

From the LNS admin UI,
I have already done the following:
1) Created an MQTT connector to talk to AWS IoT
2) Create a Handler and link it to the MQTT connector
3) Indicated the corresponding "Group" for the registered Device and Node.

May I double check with you on the following to see if I understood correctly or not:
1) In the MQTT connector, for the Client ID, is this referring to the 12-digit AWS account ID?
2) What username and password should be used for the "Name" and "Password" (the main AWS account log in credentials, or one corresponding to a created "User" in IAM)? Or should it be left blank?
3) For user.cert and private.key, i have added for a "Thing" being created in my AWS IoT. Is this correct?
4) In the Device (and Node), "backend" has been selected for "Application" type.
5) In the Device item, I have left the "Arguments" blank.
6) In the Handler, I have selected "JSON", and inserted this as the uplink function (just for test purpose):

fun(_Port, <<Count, Data:Count/binary>>) ->
  #{message => binary_to_list(Data)}
end.

Thanks a lot in advance for your pointers and guidance!

Debug Logs:
2017-06-12 17:48:28.929 [info] <0.254.0> server started
2017-06-12 17:48:28.958 [debug] <0.260.0>@lorawan_connector_mqtt:init:23 Connecting aws-iot to mqtt://a3inov2csy96yo.iot.ap-southeast-1.amazonaws.com:8883
2017-06-12 17:48:30.287 [debug] <0.260.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt311 {shutdown,tcp_closed} reconnect 0
2017-06-12 17:48:30.378 [debug] <0.260.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt31 {shutdown,tcp_closed} reconnect 1
2017-06-12 17:48:35.558 [debug] <0.260.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt311 {shutdown,tcp_closed} reconnect 2
2017-06-12 17:48:36.240 [info] <0.256.0> gateway 00800000A0000B87 {connected,{{192,168,2,1},48420}}
2017-06-12 17:48:45.285 [info] <0.269.0>@lorawan_mac:handle_join:166 JOIN REQUEST <<183,59,25,33,169,154,6,154>> <<0,128,0,0,0,0,215,231>> -> <<87,152,49,11>>
2017-06-12 17:48:45.288 [debug] <0.269.0>@lorawan_mac:handle_join:185 Join-Accept in RX1: {0,8,923.3}
2017-06-12 17:48:45.652 [debug] <0.260.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt31 {shutdown,tcp_closed} reconnect 3
2017-06-12 17:48:55.848 [debug] <0.272.0>@lorawan_mac:handle_uplink:360 DataRate <<87,152,49,11>> switched to dr 1
2017-06-12 17:48:55.853 [debug] <0.272.0>@lorawan_mac_commands:build_fopts:32 <<87,152,49,11>> <- [dev_status_req]
2017-06-12 17:49:00.766 [debug] <0.260.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt311 {shutdown,tcp_closed} reconnect 4
2017-06-12 17:49:20.927 [debug] <0.260.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt31 {shutdown,tcp_closed} reconnect 5
2017-06-12 17:49:28.427 [debug] <0.299.0>@lorawan_mac_commands:handle:17 <<87,152,49,11>> -> [{dev_status_ans,255,7}]
2017-06-12 17:49:28.427 [debug] <0.299.0>@lorawan_mac_commands:handle_status:131 DevStatus: battery 255, margin: -25
2017-06-12 17:49:46.125 [debug] <0.260.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt311 {shutdown,tcp_closed} reconnect 6

gotthardp, 2017-06-12:
Hello. Have you also created the AWS policy, enabled it and assigned the device to that? See https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md.

The username and password are empty, but you probably misconfigured something on the AWS side.

frozzygoh, 2017-06-12:
Hi Gotthard,

Yes, I have created the AWS policy as follows:

{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": [
        "*"
      ]
    }
  ]
}

Do you see any other parts that are worth checking again?

gotthardp, 2017-06-12:
The policy is different to what I used. Are you sure it is correct? I had:
```
"Statement": [{
  "Action": "iot:*",
  "Resource": "*",
  "Effect": "Allow"
}]
```

frozzygoh, 2017-06-13:
I'm sorry, the special characters (asterisk) disappeared because of markdown - have updated it above.
It looks like the difference in on the square brackets for Resource value. I will try again with the one you use this time round, and observe again. :)

Thank you Gotthard!

gotthardp, 2017-06-13:
I suppose there is no error log from AWS that would indicate something?

frozzygoh, 2017-06-13:
Yes, there is no error log from AWS indicating any sign of incoming MQTT messages received.
Still trying to figure out where I have configured something incorrectly or missed out. :P

frozzygoh, 2017-06-14:
Hi Gotthard,

I have updated the AWS policy for the AWS "thing". 

in the lora network server debug.logs, I am still getting this:
2017-06-14 12:35:41.280 [debug] <0.266.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt31 {shutdown,tcp_closed} reconnect 11
2017-06-14 12:36:11.448 [debug] <0.266.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt311 {shutdown,tcp_closed} reconnect 12
2017-06-14 12:36:41.604 [debug] <0.266.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt31 {shutdown,tcp_closed} reconnect 13
2017-06-14 12:37:11.758 [debug] <0.266.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt311 {shutdown,tcp_closed} reconnect 14
2017-06-14 12:37:41.862 [debug] <0.266.0>@lorawan_connector_mqtt:handle_reconnect0:120 Connector attempt31 {shutdown,tcp_closed} reconnect 15

and the warnings 

[warning] [Client <0.418.0>] Connection lost for: tcp_closed
[warning] [Client <0.420.0>] Connection lost for: tcp_closed
[warning] [Client <0.437.0>] Connection lost for: tcp_closed
[warning] [Client <0.440.0>] Connection lost for: tcp_closed

Do you have any suggestions on how to overcome this?

frozzygoh, 2017-06-14:
Hi Gotthard,

I am finally able to connect to AWS IoT with the Backend MQTT Connector! :)

So the main problem I found causing the connectivity issue was:
1) in URI: Missing the "s" after mqtt. Should be mqtts instead of mqtt.
2) Client ID: It should correspond to the name of the "Thing", instead of "AWS Account ID".

Phew.....

Anws, I appreciate your prompt responses in supporting earlier. Thanks a lot! The LNS rocks!

gotthardp, 2017-06-14:
Oh, I have missed the missing "s". Sorry for that.

The clientID issue is a bit weird because I was using the Client ID (a number), not the thing name. In fact AWS allows to have one MQTT connection for multiple things, so putting the thing name into the clientID doesn't make sense.

Please note that I **did not** create a device certificate. I created a general certificate and then attached a device to that. Could this be the difference?

frozzygoh, 2017-06-14:
Oh I see, that is interesting. I have been using the device certificate attached to the device or "thing", so I think both options are also possible for the secure connection to take place. 

Thank you so much for the insights!

closed 2017-06-14 by frozzygoh