gotthardp/lorawan-server #127: {uplinks_lost,58}

blackansible, 2017-06-12:
I tried to connect a node using ABP with my Gateway and i can see traffic {uplinks_lost,58} but i dont know how next steps i need to do to process the info. Do i need to create connectors and handlers? The documentation is not to clear in this point. I am not expert in Lora.

gotthardp, 2017-06-12:
The `{uplinks_lost,58}` could just mean the server/gateway was off while the device was transmitting data. This is not necessarily an issue.

What are you trying to achieve? How do you want to further process the data?

blackansible, 2017-06-12:
Dear Petr @gotthardp

As i told you, i am trying to use your lorawan-server with lopy development board. 
https://www.pycom.io/product/lopy/

I am using the following code for the gateway

https://github.com/SeaSlugLabs/Lopy-Nanogatway-USA/blob/master/Nanogateway-Code/lib/nanogateway.py

And for the node

https://github.com/SeaSlugLabs/Lopy-Nanogatway-USA/blob/master/TTN-Node-Lopy-Code/main.py

The main intention is to use your server as network server for authentication and then get the info and pass to mqtt server.

In the point i am working i want to make a functional prototype. I can see the gateway alive. Is there a way to see traffic that is receiving the lora server to discard node-gateway problems?

Best regards,
Jorge Reyes

gotthardp, 2017-06-12:
You didn't mention lopy before.

The traffic received by the server can be seen on the 'Received Frames' page. That's a good start.

Once you are able to see data in 'Received Frames', you should create a Backend Connector if you want to use MQTT.

gotthardp, 2017-06-21:
Does the issue persist? May I close this?

blackansible, 2017-06-21:
Yes, Petr.

 

Close it. I will use another gateway.

 

Best regards,

Jorge Reyes