gotthardp/lorawan-server #128: Server Performance Simulation

kammyy, 2017-06-13:
Hi Petr,

I am able to successfully connect a few gateways and few end nodes with the server. Till now the performance is pretty good.

But what would happen if 100s of gateways and 1000s of end nodes are connected to it ?

Is there way to simulate the server performance using some software code or some other tool ?

One option could be sending lots of UDP packets to the server just as a gateway sends. 

Please suggest a solution.

gotthardp, 2017-06-13:
Hi. You could extend the test tools to simulate gateways and nodes. Currently the tests are run on the same machine. Alternatively one could use these modules to develop a load-test tool and run that from another machine.

kammyy, 2017-06-13:
What are these test tools and how to develop a load-test ?

Thanks

kazi93, 2017-06-13:
@kammyy @gotthardp I would like to run that from another machine. Is there are any available documentation? @kammyy Can we use Lorasim or lpwansim to send such massive number of UDP to lorawan-server? Many thanks.

kammyy, 2017-06-13:
@kazi93 : we currently dont have such a tool. We are also looking for it.
Will share with you when i found/develop one such tool.

gotthardp, 2017-06-13:
@kazi93 @kammyy I have a packet_forwarder simulator and a LoRa-mote simulator already (in https://github.com/gotthardp/lorawan-server/tree/master/test) and I am about to extend it towards load-testing, but it will take me few days.

gotthardp, 2017-06-13:
I don't think the lorasim nor lpwansim is what you're looking for.

kammyy, 2017-06-13:
@gotthardp : It will be really helpful to have the packet_forwarder simulator. Will be waiting for you to complete the module.

Thanks

gotthardp, 2017-06-13:
@kammyy Do you have any wishlist of what features you'd need? (I don't say I implement it all.)

I would need just a simple load-tester to send a high amount frames towards an application (likely the semtech mote), simulating traffic from multiple gateways and multiple devices.

kammyy, 2017-06-15:
@gotthardp : Thanks for asking. I will be waiting for you to compete the testing module. And will be sharing the wishlist soon :P

siscia, 2017-07-01:
This: https://gitlab.com/itk.fr/lorhammer

May fulfil some of your needs :)

gotthardp, 2017-07-02:
Oh, this looks like the right tool. Thanks for letting us know!
@kazi93 @kammyy, will you please try to run the tool against your lorawan-server installations?

closed 2017-07-15 by gotthardp