gotthardp/lorawan-server #132: mqtt downlink

YousseMes, 2017-06-16:
can you please tell me how i can send a downlink , I use mosquitto it work good by testinf sub and pub 
when i send data i have this problem :/

[image]
[image]
[image]
[image]

YousseMes, 2017-06-16:
also I have a message in server lorawan [warning] ClientID is NULL!

gotthardp, 2017-06-16:
Could you try removing the (unnecessary) spaces between JSON fields?

The `[warning] ClientID is NULL!` is harmless. You can fix it by specifying some (whatever) ClientID for the connector.

YousseMes, 2017-06-19:
yes I made it but i can't always make a downlink :/

YousseMes, 2017-06-19:
[image]

i see that it was a downlink fonction but in the node nothing 
[image]

gotthardp, 2017-06-19:
what exactly was the command you used? the devaddr has to match, obviously

YousseMes, 2017-06-19:
mosquitto_pub -h 127.0.0.1 -p 1883 -t 'in/260118B3' -m '{"devaddr":"260118B3","data":"00","confirmed":true,"port":1,"time":"immediately"}'

gotthardp, 2017-06-19:
try without "time: immediately" please

YousseMes, 2017-06-19:
always the same problem  the message sent but i can't see downlink in the node 
please how can i see the log ?

gotthardp, 2017-06-19:
https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md

YousseMes, 2017-06-19:
[image]
also i have this problem

YousseMes, 2017-06-19:
i have this problem 
2017-06-19 11:50:29.083 [error] <0.263.0>@lorawan_application_backend:data_to_fields:203 Fun failed error:function_clause

gotthardp, 2017-06-19:
This means the erlang functions for uplink/downlink parsing/building are wrong. If you post them here I can check them for you.

YousseMes, 2017-06-19:
[image]

gotthardp, 2017-06-19:
And what data is the device sending?, What is on the Received Frames list?

YousseMes, 2017-06-19:
[image]

YousseMes, 2017-06-19:
it's some data from arduino lmic it's in hexadecimal

gotthardp, 2017-06-19:
But the uplink parser is designed for data from Semtech Mote, which sends much more data. You have to change them, or simply remove them all and keep the field blank.

YousseMes, 2017-06-19:
so how can i send a downlink ?

YousseMes, 2017-06-20:
ok it's work now i used rabbitmq i can send a downlink 
i close that

closed 2017-06-20 by YousseMes