gotthardp/lorawan-server #133: Fcnt Check Value: Strict 16 or Strict 32 or Restart on zero or Disable ?

kammyy, 2017-06-19:
Hi Petr,

I have been testing the server with around 50 devices. I am using ABP join method. What should i keep  the value of Fcnt Check ?

Case 1: When i keep it Strict 16 or strict 32 then if the device is rebooted at some time and the device resets the uplink fcnt, in this case i start to get "fcnt gap too large error"

e.g. if uplink fcnt = 100 and the device reboots then till the time the uplink fcnt again reaches 100 i keep on getting fcnt gap too large error.

Case 2: When i keep it to Reset on Zero . In this case if the device reboots and after reboot the very first message with upfcnt "0" reaches the server. Everythings works fine. But if the very first message with fcnt zero does not reaches the server then i keep on getting the "fcnt gap too large" till the uplink fcnt reaches the previuos session uplink fcnt value.

Case 3: When i keep it to disable. In this case, there is no check so its not option to use.

According to me, When we select Strict 16 or strict 32 or Reset on zero : in all these cases if the server receives a fcnt which is less than the value of fcnt in the previous session (it can be zero or any other number less than the previous session value), the server should detect this as a new session and accept the message without categorizing it as a "fcnt gap too large error".

Please share your views/thought process on this.
.

gotthardp, 2017-06-19:
Hello. Your observations are correct. In case of "Reset on Zero" there is a small interval of 10 frames (0 to 9) that are accepted as reset. If the device reset and then 11 frames are lost, it produces the "fcnt gap too large" error.
All of this is due to protection from reply attacks. If the server would accept any smaller packet as a reset, it would be too easy for hackers to send a reply attack. Therefore, only 10 packets are allowed.

kammyy, 2017-06-19:
Can we have a configurable number of allowed fcnts ? e.g. like we have 10 number of allowed fcnts. It would be great to have that number configurable.

Thnks

gotthardp, 2017-06-19:
Can be done.

gotthardp, 2017-06-21:
The latest `master` allows you to change this in the sys.conf using the `{max_lost_after_reset, 10},` config entry. Please try.

closed 2017-06-28 by gotthardp