gotthardp/lorawan-server #134: Compiling on Raspbian

kwmartin, 2017-06-26:
Just a comment to save others some time: As Petr clearly states: You will need the Erlang/OTP 19 or later. Unfortunately, Raspbian's Erlang/OTP is not version 19. One needs to upgrade Erlang using Backports. That is:
1) add deb http://ftp.debian.org/debian jessie-backports main to /etc/apt/sources.list (or add a new file to with the ".list" extension to /etc/apt/sources.list.d/)
2) run sudo apt-get update
3) run
>sudo apt-get -t jessie-backports purge erlang
and then
>sudo apt-get -t jessie-backports install erlang
after which running
>make release
should compile correctly.

gotthardp, 2017-06-26:
Thanks for the information. Alternatively one could also try downloading Erlang 20 for Raspbian from https://www.erlang-solutions.com/resources/download.html

This information is included in the Installation guide, but I may need to mention that more often.

kwmartin, 2017-06-27:
Petr, I went to https://www.erlang-solutions.com/resources/download.html as suggested and followed the instructions there (to the best of my ability). I first did 1) and then did the alternative method. In both cases, after running >sudo apt-get update, and then >sudo apt-get install erlang it stated my version (version 19) was the latest version. Is it possible that version 20 has not yet been ported to raspbian? Or am I missing something obvious?

gotthardp, 2017-06-27:
But Erlang 19 is also fine. You don't need to struggle obtaining R20.

closed 2017-07-02 by gotthardp