gotthardp/lorawan-server #135: Could not start kernel pid

mrdloof, 2017-06-26:
Hello,
I'm working with openWRT and I'm trying to run the server. We've followed the instructions but when we try to run it with ./lorawan_server we get a strange error. 

2017-06-26 16:22:33 Error in process ~p with exit value:~n~p~n
        <0.31.0>
        {{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,"application_controller.erl"},{line,1809}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]},{application_controller,check_conf,0,[{file,"application_controller.erl"},{line,1808}]},{application_controller,init,2,[{file,"application_controller.erl"},{line,485}]}]}}},[{application_controller,init,2,[{file,"application_controller.erl"},{line,485}]}]}
{"could not start kernel pid",application_controller,"{{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,\"application_controller.erl\"},{line,1809}]},{lists,foldl,3,[{file,\"lists.erl\"},{line,1263}]},{application_controller,check_conf,0,[{file,\"application_controller.erl\"},{line,1808}]},{application_controller,init,2,[{file,\"application_controller.erl\"},{line,485}]}]}}},[{application_controller,init,2,[{file,\"application_controller.erl\"},{line,485}]}]}"}
could not start kernel pid (application_controller) ({{case_clause,{'EXIT',{function_clause,[{application_controller,'-check_conf/0-fun-0-',[[],[]],[{file,"application_controller.erl"},{line,1809}]},{

Crash dump is being written to: erl_crash.dump...done

Any idea how to fix this?
Thank you in advance

gotthardp, 2017-06-26:
But the instructions say you should run the server `bin/lorawan-server`. ;-)

What Erlang do you use? How did you install Erlang and from what source?
What server version are you using?

mrdloof, 2017-06-26:
We are on a limited environment with low storage. We compiled erlang OTP 19.2 from source and compiled the source with c(lorawan_server).

Make is not available on our platform. The executable is thus perfectly created without compile errors.
There is no bin folder in the project so I assumed that that was the way to do it :) I made a bin folder and put the scripts in it. It gives the same error.

We also tried to place the server config on many places.

gotthardp, 2017-06-26:
You need the rebar3 tool to compile it. `c(lorawan_server)` doesn't do the job. The `bin` directory is auto-created during compilation.

mrdloof, 2017-06-27:
We installed Erlang 20.0 and rebar3. Compiling with rebar works fine and we get the bin folder in _build/default/rel. But when we do bin/lorawan-server we get this error: 

"init terminating in do_boot()"

any idea of what's wrong?

mrdloof, 2017-06-27:
We were compiling with Erlang 20 but the version on our openWrt was Erlang 19.2 . We changed the version on openWrt to 20.0 and it worked! :)

closed 2017-06-27 by mrdloof