gotthardp/lorawan-server #137: CFList on Join Accept message

julien78910, 2017-06-28:
Hello gotthardp,

Sorry to bother you again but I have a question on a feature that I didn't found on this project.
It is the CFList bytes in the Join Accept message to configure the Tx channels of the LoRa modules after an OTAA.
Is it a feature that already exist or should I wait for a next version ?

I can pre-configure my LoRa modules so it is not a big deal if this feature doesn't exist. Just if it is already done, I would use it to have something more configurable.

Thanks,
Julien

gotthardp, 2017-06-28:
Hi Julien, the CFList is not implemented. I will implement it in some of the future releases.

gotthardp, 2017-12-28:
CFList implemented; will be in the future 0.5.0 release

closed 2017-12-28 by gotthardp