gotthardp/lorawan-server #138: Can't add a gateway after installing a server with docker

Suko, 2017-06-29:
Maybe I am missing something but I can't add a gateway after installing a server with docker tag "latest". I added Google Maps API key and map is shown but gateway still can't be added. This is an error message from the console.

TypeError: Cannot read property 'length' of null
    at Array.parse_bitstring (admin.js:793)
    at e.value (ng-admin.min.js:1)
    at ng-admin.min.js:1
    at Array.forEach (<anonymous>)
    at e.value (ng-admin.min.js:1)
    at e.value (ng-admin.min.js:47)
    at fn (eval at compile (ng-admin.min.js:9), <anonymous>:4:284)
    at i (ng-admin.min.js:10)
    at p.$eval (ng-admin.min.js:8)
    at p.$apply (ng-admin.min.js:8)

Suko, 2017-06-29:
The same problem occurs with Docker stabile.
 
This is on location admin.js:793
function parse_bitstring(value, entry) {
    if(value.length > 0)
    {
        var parts = value.split(':', 2);
        return {val: parts[0], len: +parts[1]};
    }
    else
        return null;
}

Suko, 2017-06-29:
Strange behavior: If I add field SubID and delete it submit works.
Posible solution:
if(value && value.length > 0)
{
var parts = value.split(':', 2);
return {val: parts[0], len: +parts[1]};
}
else
return null;
}

gotthardp, 2017-06-29:
Oh, that's a bug. I am sorry for that.
It's been there for a while. Thanks for reporting this, I just committed the fix you suggested.

gotthardp, 2017-07-05:
Fixed in v0.4.6

closed 2017-07-05 by gotthardp