gotthardp/lorawan-server #139: Possible to add device with empty fields with REST API

Sviests, 2017-06-29:
Hi! I managed to add a device using the rest api without filling any fields. What I now have is a device showing without any fields and without a possibility to edit it or even delete it!

[image]

gotthardp, 2017-06-29:
Interesting :) Could you tell me what HTTP DELETE request is sent when you try to delete that row? (You can check this in Mozilla via the Developer menu -- Network.

Sviests, 2017-06-30:
I see what the problem is, it is trying to send a DELETE request to /devices/ with an empty DevEUI and the allowed methods for /devices are only GET, POST and OPTIONS. So the response to the DELETE request is a 405 Method Not Allowed.

gotthardp, 2017-06-30:
Allright, you need to modify the database directly.

Connect to a running lorawan-server by `erl -sname test -remsh lorawan@localhost`, where instead of `localhost` use the short hostname of the machine hosting the server (`hostname --short`).

Then delete the entry by `mnesia:dirty_delete(devices, undefined)`, which should return `ok`.

Then close the remote shell and you're done.

Sviests, 2017-06-30:
Ok, thanks for the instructions!

gotthardp, 2017-06-30:
Please let me know if/how it worked.

gotthardp, 2017-07-05:
Fixed in v0.4.6

closed 2017-07-05 by gotthardp