gotthardp/lorawan-server #140: Server crash: "reason bad argument in gen_fsm:send_event/2 line 215"

dnouri, 2017-06-29:
We suspect the reason for our lorawan-server crash here was that our mqtt broker went down for a while.

I'm attaching logs from the crash.  We're running version 0.4.3, upgraded because we wanted the fix for this: https://github.com/gotthardp/lorawan-server/commit/22c4d21d19d0e868a1a43dcb175d2eded105f985

[crash.log.txt](https://github.com/gotthardp/lorawan-server/files/1112550/crash.log.txt)
[debug.log.txt](https://github.com/gotthardp/lorawan-server/files/1112549/debug.log.txt)
[error.log.txt](https://github.com/gotthardp/lorawan-server/files/1112551/error.log.txt)

gotthardp, 2017-06-29:
I guess this because a frame arrived, which had to be published, while the client has been reconnecting. This is of course a bug, whcih I will fix. Thanks for reporting this.

gotthardp, 2017-07-12:
Would you please try v0.4.6 to check if the issue is fixed?

dnouri, 2017-07-18:
@gotthardp Thanks a lot for the fix, and sorry for not replying earlier.  It looks like your change fixed the issue.  Although admittedly we had issues reproducing it in the first place.

closed 2017-07-18 by dnouri