gotthardp/lorawan-server #141: unknown error, unable to send downlink

Ankitjd18, 2017-06-29:
Hi,
We are testing the server and found some error log 
`2017-06-29 16:32:19.288 [debug] <0.1370.0>@lorawan_mac:check_link_fcnt:274 <<112,0,208,43>> fcnt reset
2017-06-29 16:32:19.288 [error] <0.1370.0> gen_server 'wpool_pool-handler_pool-1' terminated with reason: no case clause matching {badmatch,{866.55,<<"SF8BW125">>}} in wpool_process:handle_cast/2 line 155
2017-06-29 16:32:19.288 [error] <0.1370.0> CRASH REPORT Process 'wpool_pool-handler_pool-1' with 0 neighbours crashed with reason: no case clause matching {badmatch,{866.55,<<"SF8BW125">>}} in wpool_process:handle_cast/2 line 155
2017-06-29 16:32:19.288 [error] <0.1332.0> Supervisor 'wpool_pool-handler_pool-process-sup' had child 'wpool_pool-handler_pool-1' started with wpool_process:start_link('wpool_pool-handler_pool-1', lorawan_worker, [], [{queue_manager,'wpool_pool-handler_pool-queue-manager'},{time_checker,'wpool_pool-handler_pool-time-checker'},...]) at <0.1370.0> exit with reason no case clause matching {badmatch,{866.55,<<"SF8BW125">>}} in wpool_process:handle_cast/2 line 155 in context child_terminated
2017-06-29 16:32:19.288 [debug] <0.1332.0> Supervisor 'wpool_pool-handler_pool-process-sup' started wpool_process:start_link('wpool_pool-handler_pool-1', lorawan_worker, [], [{queue_manager,'wpool_pool-handler_pool-queue-manager'},{time_checker,'wpool_pool-handler_pool-time-checker'},...]) at pid <0.1371.0>`
We are unable to understand what this is about, but downlink stopped going. Can you please help us here.
Thanks in advance.

gotthardp, 2017-06-29:
Hello. In your server config file you have `{866.55,<<"SF8BW125">>}`. The format changed recently, so please change that to `{866.55,<<"SF8BW125">>,<<"4/5">>}`.

Ankitjd18, 2017-06-30:
I made the changes, the error is gone now. But, unable to send downlink, also ack to uplink is also not going.

gotthardp, 2017-07-01:
Could you please provide more details? Is there some error printed?

Ankitjd18, 2017-07-01:
There doesn't seem to be much error, only found below in logs -
`2017-07-01 11:22:51.950 [error] <0.264.0> gateway 00800000A00000B7 <<67,79,76,76,73,83,73,79,78,95,80,65,67,75,69,84>>
2017-07-01 11:22:57.903 [warning] <0.264.0> gateway 00800000A00000B7 {downlinks_lost,1}`

Ankitjd18, 2017-07-01:
`2017-07-01 13:17:27.207 [warning] <0.256.0> node 7000D003 downlink_lost
2017-07-01 13:17:27.207 [debug] <0.256.0>@lorawan_mac:handle_uplink:422 <<112,0,208,3>> retransmitting
2017-07-01 13:17:28.995 [warning] <0.250.0> gateway 00800000A00000B7 {downlinks_lost,1}`
getting this when trying to send downlink, and uplink also stops coming after this. If we don't send downlink, ack is working fine for uplink.

Ankitjd18, 2017-07-12:
Thanks for your help. There was some issue with the device. It's working fine.

closed 2017-07-12 by Ankitjd18