gotthardp/lorawan-server #142: Billing add-on

batulzii, 2017-07-01:
This server is private LoraWAN network, and it is solid enough to use for commercial purpose.
Any plan or advice third party billing add-on?

gotthardp, 2017-07-01:
Hello. Billing is usually done by dedicated platforms based on "charging tickets" indicating that someone has consumed a certain service amount. I am not aware of any suitable open source billing system, but when you find one it should be relatively easy to generate charging tickets using any API the system supports.

batulzii, 2017-07-01:
Thank you! Petr

batulzii, 2017-07-02:
.
Could you advice how tire node with payment. Any flag that will be used for this purpose. Disable/ Enable

gotthardp, 2017-07-02:
What should the flag do? I think the server should generate little JSON messages when a frame a received and another system shall process these "charging tickets" and generate invoices.

batulzii, 2017-07-02:
got it! Thanks Petr.

batulzii, 2017-07-02:
idea about flag was in case of customer did not pay and have to disable that node

gotthardp, 2017-07-02:
The server has an ignore-list (under Infrastructure). You could put the device in that list.

closed 2017-07-02 by batulzii