gotthardp/lorawan-server #143: Docker setup will not work if you change ports in the setup process. Help/improvement needed.

Suko, 2017-07-04:
When you install the server from the docker it doesn't allow you to setup custom ports. For example, when I want to set udp port 1700, if I set them in the docker run command it will not work as initial docker setup file has set to expose port 1680, not 1700. 

Is there a way that we could expose ports which we want to use the run command for ex. I would like to expose ports to reach MQTT server 1883, 8883, ...?  

Ex. run command:
docker run --detach \
  --name lorawan \
  --hostname lorawan \
  --rm \
  --volume /data/lorawanserver:/storage \
  --publish 8080:8080/tcp \
  --publish 1700:1700/udp \

Docker install document
# data from port_forwarders
EXPOSE 1680/udp

gotthardp, 2017-07-04:
It shall work. The syntax is `--publish hostPort:containerPort`, so you should be able to change *the first* number to change the exposed port. But I don't think you need to expose anything for MQTT.

Suko, 2017-07-04:
Thank you for your explanation. I tested with MQTT and it worked too.

closed 2017-07-04 by Suko